tum-vision/lsd_slam #53: Trying to compile on Jetson TK1

MarcGyongyosi, 2015-01-06:
Hi!

So I am trying to compile LSD-SLAM on my Jetson TK1, however I ran into an issue while running rosmake.

First, I was getting an error message telling me that a ';' was expected in Frame.cpp around line 530 (in the lsd_slam_core/src/Datastructures directory). As I was looking through the code I saw that this part was only called if -DENABLE_SSE is set. Otherwise it would go to the next part if -DENABLE_NEON is set. Curious as to whether this would make it work (I must honestly admit that I do not know what I was actually changing) I changed -DENABLE_SSE to -DENABLE_NEON in the CMakeLists.txt. After calling rosmake lsd_slam again, it looked like it worked - this time it ran longer than before. Now, however, I am getting these error messages from the assembler:

  /tmp/ccnTsMpJ.s: Assembler messages:
  /tmp/ccnTsMpJ.s:10805: Error: VFP single precision register expected -- `vldmia r0,{q10}'
  /tmp/ccnTsMpJ.s:10811: Error: selected FPU does not support instruction -- `vadd.f32 q0,q0,q2'
  /tmp/ccnTsMpJ.s:10812: Error: selected FPU does not support instruction -- `vadd.f32 q1,q1,q3'
  /tmp/ccnTsMpJ.s:10813: Error: selected processor does not support Thumb mode `vpadd.f32 d0,d0,d1'
  /tmp/ccnTsMpJ.s:10814: Error: selected processor does not support Thumb mode `vpadd.f32 d1,d2,d3'
  /tmp/ccnTsMpJ.s:10815: Error: selected FPU does not support instruction -- `vmul.f32 q0,q0,q10'
  make[3]: *** [CMakeFiles/lsdslam.dir/src/DataStructures/Frame.cpp.o] Error 1

This looks like the compiler generated code that the hardware doesn't support? Please let me know if there's anything I could try to make this work!

Thanks a lot,
Marc

MarcGyongyosi, 2015-01-06:
Hi, I figured it out: I uncommented the line add_definitions("-DENABLE_SSE") and rosmake worked properly. Looks like the Jetson TK1 does not support either SSE or NEON...

closed 2015-01-06 by MarcGyongyosi

tngan, 2016-06-26:
@MarcGyongyosi I experience the same issue even remove that line. Did you do other things to make it work ?