tum-vision/lsd_slam #58: Tracking Lost problems

romulogcerqueira, 2015-01-08:
Hi everybody,

I'm using Kinect RGB Mono as live camera in LSD-SLAM and I have found this error:

$ rosrun lsd_slam_viewer viewer
$ rosrun lsd_slam_core live_slam /image:=/camera/rgb/image_mono _calib:=calib_kinect.cfg

TRACKING LOST for frame 218 (10,69% good Points, which is 48,68% of available points, NOT DIVERGED)!

My calibration file:
520.323830 521.287893 323.751970 275.976601 0.171381 -0.309677 0.004476 0.004649
640 480
full
640 480

How can I solve this problem?

Thanks.

mp3guy, 2015-01-22:
Try this calibration;

0.825 1.1 0.5 0.5 0
640 480
none
640 480

amiltonwong, 2015-01-22:
@mp3guy,
Why the values you provided : 0.825 1.1 0.5 0.5 0
differ significantly from the ones from @romulogcerqueira : 520.323830 521.287893 323.751970 275.976601 ?

I'm also puzzled in this issue as in my previous post #63

ThX~
Milton

mp3guy, 2015-01-22:
It's the ATAN camera model with no radial distortion;

fx/width fy/height cx/width cy/height 0
width height
none
width height

amiltonwong, 2015-01-22:
Thank you so much, @mp3guy 
You had relieved my trouble :)

liqile, 2015-03-24:
@romulogcerqueira hello, I am trying to use the rgb-dataset from http://vision.in.tum.de/data/datasets/rgbd-dataset. may i ask you how to get the calibration parameters..thanks a lot

yuanshucong, 2016-02-29:
@amiltonwong 
can you tell me how does the calib provided by mp3guy work ? because i used a webcam and the result is bad

yuanshucong, 2016-02-29:
@mp3guy thank you 
with your calib file   .the result is fine now

vacky11, 2017-04-17:
@mp3guy 
I am facing the same problem. But Im using iphone 6s for taking video from Iphone 6s

My calibration file is: 
1.0154 1.8441 0.5001 0.5722 -0.0005
1920 1080
crop
640 480

I got the above values using PTAM camera calibrator 
Where am I going wrong?