tum-vision/lsd_slam #63: question about calibration file

amiltonwong, 2015-01-20:
Dear all,

Take example as Desk Sequence (png file and bag file) for testing. 
For png images, the focal length and principle setting and distortion setting of cameraCalibration.cfg is:
0.771557 1.368560 0.552779 0.444056 1.156010 

While for /camera_info within LSD_room.bag file are:
K[0]: 254.326954
K[1]: 0
K[2]: 267.381897
K[3]:  0
K[4]: 375.934381
K[5]: 231.599091
K[6], K[7]: 0
K[8]: 1
That means: fx=254.326954, fy=375.934381, cx= 267.381897,cy= 231.599091.

Why are these two K settings so different significantly?

THX~
Milton

budhi15, 2015-01-21:
My guess is that they used separate cameras?

hiankun, 2015-01-22:
Also, the values in cameraCalibration.cfg might have been devided by camera width and height.

amiltonwong, 2015-01-22:
@budhi15,
Maybe. But I choose the same image source (which I extract from LSD_room.bag file) and apply them in two approaches, two results are almost the same:
[1]
rosbag play /root/Downloads/LSD-SLAM/Desk_Sequence/LSD_room.bag
rosrun lsd_slam_core live_slam image:=/image_raw camera_info:=/camera_info

or
[2]
rosrun lsd_slam_core dataset _files:=/root/images _hz:=0 _calib:=/root/Downloads/LSD-SLAM/Desk_Sequence/LSD_room/cameraCalibration.cfg

So , these two K settings(Camera matrix) which  differ significantly may have some relation between  themselves, but I can figure out the conversion formula between them. As suggest from @hiankun, still I cannot find the exact conversion formula.

Hopefully someone could point me to the answer :)

Thanks a lot~

himanshub43, 2016-07-26:
I guess its in atan model. But still the values dont match

xslittlegrass, 2018-08-11:
I did a similar test and the parameters seem to match for me.

The "camera_info" topic of LSD_room.bag has the following message:

```
header: 
  seq: 2743
  stamp: 
    secs: 1400002262
    nsecs: 209756768
  frame_id: /camera
height: 480
width: 640
distortion_model: ''
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [254.32695388793945, 0.0, 267.38189697265625, 0.0, 375.93438148498535, 231.59909057617188, 0.0, 0.0, 1.0]
R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
P: [254.32695388793945, 0.0, 267.38189697265625, 0.0, 0.0, 375.93438148498535, 231.59909057617188, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```

This means
```
fx=254.326954
fy=375.934381
cx= 267.381897
cy= 231.599091
width = 640 
height = 480 
```

and the distortion parameters are all zero.

Since the images are already rectified, we can use the calibration format corresponding to the pre-rectified images, which has the format 

```
fx/width fy/height cx/width cy/height 0
width height
none
width height
```

This means the calibration parameters are:

```
0.397386 0.783197 0.417784 0.482498 0
640 480
none
640 480
```

I use these parameters and it works well with the extracted image from the LSD_room.bag file

`rosrun lsd_slam_core dataset _files:=/Downloads/LSD_room_bag/images _hz:=0 _calib:=/Downloads/LSD_room_bag/calib.cfg`