tum-vision/lsd_slam #64: PointCloud Viewer window shows only the moving camera but no point cloud

hiankun, 2015-01-22:
Hi, I have tried to run lsd_slam for several days but still got no successful result, and I need some suggestions to find out where the problem is.

The issue I've encountered is that the code seemed keeping running (and it also lost tracking frequently, but it's might be another issue), and **the moving camera was shown in the `PointCloud Viewer` window. However, no point cloud was shown.** The following screenshot shows my result (I changed the code to alter the camera line color):

[image]

Even when I followed the [Quickstart instruction](https://github.com/tum-vision/lsd_slam#1-quickstart--minimal-setup) to run lsd_slam with LSD_room.bag file, the results were all the same (it lost tracking at random frame, around after 1000th frame).

I traced the source code with `printf` and found the program always touched the following part in `KeyFrameDisplay::drawPC()`:
```
if (! vertexBufferIdValid )
{
    return ;
}
```

Also, when I pressed `Ctrl-w` on `PointCloud Viewer` window when lsd_slam was running with LSD_room.bag, the terminal always returned
```
Have 0 points, 0 keyframes, 0 constraints. Displaying 0 points.
```

So the program generated no point cloud, even when the tracking was still working?

At least I should get some reasonable results with the LSD_room.bag example, but I didn't. Did I miss something essential? I need some hints and helps from you. Thank you.

ghemann, 2015-03-30:
Did you find the reason for this?  I'm having the same issue as well.
Thanks

hiankun, 2015-07-13:
I have left this topic for a long time... I would retry the code if I have some leisure time.

Thank you for the reply. :-)

gitunit, 2015-09-14:
@ghemann can you tell me please where did you make the changes?

ghemann, 2015-09-14:
lsd_slam/lsd_slam_core/src/IOWrapper/OpenCV/ImageDisplay_OpenCV.cpp
lines 119 and 125
I just commented them out and put 'return 1'.

gitunit, 2015-09-14:
@ghemann great, it finally works! thx. i should have asked earlier ;)

hiankun, 2015-10-01:
@ghemann It works! Thank you. :-)

The viewer now shows the point cloud with the moving camera.
[image]

closed 2015-10-01 by hiankun

yuanshucong, 2016-02-29:
Hi @hiankun  
Can you tell me the camera you used and the lens? Did you use a global shutter camera?

hiankun, 2016-06-07:
@yuanshucong Sorry for the so-late reply. The camera I used here is the Genius WideCam 1050. FYI.