tum-vision/lsd_slam #65: How to get estimated camera pose

fengcs, 2015-01-26:
I install the program successfully, and run the test dataset file successfully,as picture shows:

[image]

However, I wander is there a way to get the estimated camera pose file(file format as  [vision group](http://vision.in.tum.de/data/datasets/rgbd-dataset ) ) ? After reading the whole readme, I don't get any clue.

henn1001, 2015-01-26:
I logged them with "rostopic echo -p  /lsd_slam/pose > test" . That should give you a csv file.

aartikanungo, 2015-10-15:
hey i got the CSV file but i want to get the image points or keyframe points or row column matrix of images through witch the camera pose file is generated. Do you have any idea, how to estimate those points?

langxm1223, 2015-10-15:
@aartikanungo , hello
I also want to log these files.
But I tried "rostopic echo -p /lsd_slam/pose > test", it returned a file without format. Is it supposed to be csv file?
The pose have all 6 coordinates for each frame.
I am not sure whether these coordinates are relative or global? 
Have you tried to save keyframes or liveframes topic? I don't know how to deal with that either.
Any idea for this?

Thank you.

aartikanungo, 2015-10-27:
hello @langxm1223 , yes that is csv file and the coordinates are relative not global. and i am  still trying to save keyframes or the points of each image through which the cloud points are generated. if you get any solution regarding to this please let me know. Thank you

apesIITM, 2017-04-16:
Hello @aartikanungo @langxm1223, I have extracted poses as mentioned above. Can you tell how to get information from lsd_slam/keyframes (such as depth map of each keyframe, camtoWorld). I have been stuck at this for some time. 
Thanks