tum-vision/lsd_slam #68: Minimal Setup Problem: Couldn't find executable named live_slam

github-tn, 2015-02-03:
Hi all,

I'm new to lsd slam and trying to get minimal setup work. However, I got error at line **rosrun lsd_slam_core live_slam image:=/image_raw camera_info:=/camera_info**, the error message is **rosrun lsd_slam_core live_slam image:=/image_raw camera_info:=/camera_info**.

Thanks in advance.

closed 2015-02-15 by github-tn

Volcanoo, 2016-02-15:
@ningt Hey Tang, 

I'm new and have the same issue. Would you like to tell me how was this problem finally solved?

Thanks a lot.

uzair789, 2016-05-12:
I managed to fix this problem by using catkin instead of rosmake. The following link provides step by step instructions:

http://visbot.blogspot.com/2014/11/tutorial-building-of-lsd-slam-on-ros.html