tum-vision/lsd_slam #75: different camera running lsd slam

snakehaihai, 2015-03-02:
Hi Jacob
I`ve test the lsd slam using quite a few camera and different lens.  The best result still the one you mentioned on your homepage UI 1221 + 2.4mm lens. For all the camera and lens. i calibrated each set with ros calibration node and ptam_calibrator to generate different calibration configuration file. 

I tested on pointgrey flea3 and grasshopper2 + fisheye 1.4 mm lens(160 degree) and also with a normal 97 degree lens. 
with fish eye lens. the camera tracking is really good. never lost a frame. but the reconstruction is very bad. In the slam viewer the reconstructed 3d pointcloud appears to be chaos. Is there any way to solve this problem?

with lesser degree lens(on cameras such as flea3, grasshopper, UI-3250cp) , the tracking lost rapidly and mapping shows in correct depth. say far away object appears to be red and closer object appear to be blue. In a word, the inverse pixel depth never converges
Tracking lost can be understand. but why even with small motion circle around a object, the inverse depth never coverage?

Regards
Shenghai Yuan

JakobEngel, 2015-03-02:
Hi Shenghai,
Hmm, I cant really say what the problem is, without seeing it. you can try adjusting the variance threshold for visualization in the viewer. What image resolution do you use? Some things in the code (like the number of pyramid levels, or the maximal length of epipolar stereo search) are hardcoded to resolutions around 640x480... but easy to change.

One more thing is that LSD-SLAM strictly enforces real-time, i.e. if you're running on a slow computer, many frames (especially for mapping) are dropped, which can be very bad.
Try running at lower speed, or on half resolution to see if that's the problem.

Jakob
Best,
Jakob

snakehaihai, 2015-03-04:
I tried to adjust the variable. but so far not give me any good result on other cameras. 
I`m aware that a lot is coded to suit the standard VGA format. So all the camera i have are VGA or similar at 60 or higher frequency. Computer wise. I`m running on alienware. so shouldn`t be a problem. 
May I ask what are the 4 parameters in the lsd-slam debug? 
Also for the 160degree lens. could it be the problem what lens is too wide that distant from center part of feature is blurred after undistort? Then certain feature just diveraged?

Also so far the mapping only can maps a few meters in front of camera. what to change in order to map  object far away?

Tgaaly, 2015-06-21:
@JakobEngel  Can you give an overview of how to do the easy changes you mentioned to configure for different resolution cameras? I'm using an ardrone front camera. 

Also, you mentioned running at a lower speed. How can this be done in the code or config files?

It would be great if you could give an overview of how to make these changes. Thanks!

JakobEngel, 2015-06-22:
Hi,
*most* settings are aggregated in settings.h / settings.cpp. I have never actually tried to run it on much larger resolutions, however we have used much smaller resolutions (down to ~160x120).
The most important settings to change (with no claim to completeness) are:

* pyramid levels
> #define SE3TRACKING_MIN_LEVEL 1
> #define SE3TRACKING_MAX_LEVEL 5
> #define SIM3TRACKING_MIN_LEVEL 1
> #define SIM3TRACKING_MAX_LEVEL 5
> #define QUICK_KF_CHECK_LVL 4
level i means down-scaled images are by a factor of 2^i. Note that the image width/height has to be a multiple of 2^PYRAMID_LEVELS. This should be set such that the coarsest resolution is around 40x30.
Also note that per default tracking is only done up to half-resolution (lvl 1) - you can set the MIN levels to 0 to get more accuracy.
Also note that some per-level thresholds are hardcoded in SlamSystem::testConstraint(...). If you disable SLAM that function is never called, otherwise it may well segfault if you change SIM3TRACKING_MIN_LEVEL / SIM3TRACKING_MAX_LEVEL. :)

* max epipolar line search length
> #define MAX_EPL_LENGTH_CROP 30.0f // maximum length of epl to search.
can be interpreted as the maximum pixel-distance something moves due to translation in between 1-2 frames. if your resolution is larger / FoV smaller, this should probably be larger. But it will make mapping slower.

* setting KFDistWeight = KFUsageWeight = 6 will take much more keyframes (-> larger map, more loop-closures -> slower), but generally give much better & more robust results.

Tgaaly, 2015-06-23:
This is great @JakobEngel - thanks a million!

Do you any advise on running it on a low FPS camera? or a rolling shutter camera? I'm getting about 30fps on my ardrone camera (over wifi). Motion blur is also a bit of a problem. I thought this was the reason the lsd-slam was not picking up a lot of KFs. I increased KFDistWeight and KFs substantially increased but no where near as many as you have in your demo video.

JakobEngel, 2015-06-23:
Yes, the amount of keyframes in the video is very high, probably way too hight - it's actually from an older version of LSD-SLAM. 
If rolling shutter is a problem, using a lower resolution may actually help, as the caused distortions measured in pixels will be less. Essentially the rolling shutter has a similar effect to a very bad intrinsic camera calibration. 
30 fps video should be absolutely fast enough, have a look at which fps tracking / mapping are actually running. What always helps is a good motion prediction model, to get better initializations for frame-to-frame tracking.

boonflies, 2015-11-02:
Jacob, I often have "TRACKING LOST" problem with lsd_slam using AR.Drone 2.0 camera. I tried to run lsd_slam slowly by moving it handheld but the same thing happens. I run lsd_slam on a pc with ubuntu 14.04, i3 and 4GB RAM but I often find the video feed from lsd_slam is not continuous. Do I face this problem because my system configuration is not sufficient enough to run lsd_slam? 

And other thing I noted is, the tracking often gets lost when I have to turn around a corner. Any suggestions please.

yuanshucong, 2016-02-25:
Hi Shenghai
       May i ask Where do you buy the lens? And how much did you buy the UI 1221?

ank700, 2016-02-26:
@snakehaihai Hi, can you please tell me what lens (model name, etc) did you use with UI 1221 LE camera and also the fisheye lens with pointgrey cameras ?

Thanks

yuanshucong, 2016-02-26:
@ank700 Hi,can you tell me how much did you get the UI 1221 LE?

ank700, 2016-02-26:
I did not buy it yet, but it should be around 300-400 Euros. I am not sure.

yuanshucong, 2016-02-26:
OK thank you. it is the same price here.
but i find some 8 Euros with the same name.but i didn't buy it because it possibly has a very poor result.

paul3333, 2021-03-02:
@Tgaaly  were you able to identify parameters that lead to a good reconstruction? would you mind sharing it with us? I am too using the ardrone and want to achieve a best possible reconstruction.