tum-vision/lsd_slam #76: Move to catkin build system

dtmoodie, 2015-03-05:
Please port to catkin build system to work with newer ROS installs.

icoderaven, 2015-03-05:
Can you please confirm that pull request #70 works for you?

colek42, 2015-03-20:
your fork works for me on 14.04, indego.  Thanks!

chrisspen, 2015-05-04:
It would be great if these were merged into master.

chrisspen, 2015-05-04:
@icoderaven, Your fork doesn't compile for me on 14.04, Indigo. I get the error:

    /home/chris/git/lsd_slam_test/src/lsd_slam/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:24:34: fatal error: g2o/core/base_vertex.h: No such file or directory
     #include "g2o/core/base_vertex.h"
                                      ^
    compilation terminated.
    make[2]: *** [lsd_slam/lsd_slam_core/CMakeFiles/lsdslam.dir/src/DataStructures/Frame.cpp.o] Error 1
    make[1]: *** [lsd_slam/lsd_slam_core/CMakeFiles/lsdslam.dir/all] Error 2
    make[1]: *** Waiting for unfinished jobs....
    [ 10%] Building CXX object lsd_slam/lsd_slam_viewer/CMakeFiles/viewer.dir/src/PointCloudViewer.o
    In file included from /home/chris/git/lsd_slam_test/src/lsd_slam/lsd_slam_viewer/src/main_viewer.cpp:25:0:
    /home/chris/git/lsd_slam_test/src/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h:30:41: fatal error: lsd_slam_viewer/keyframeMsg.h: No such file or directory
     #include "lsd_slam_viewer/keyframeMsg.h"
                                             ^
    compilation terminated.
    In file included from /home/chris/git/lsd_slam_test/src/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.cpp:22:0:
    /home/chris/git/lsd_slam_test/src/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h:30:41: fatal error: lsd_slam_viewer/keyframeMsg.h: No such file or directory
     #include "lsd_slam_viewer/keyframeMsg.h"

icoderaven, 2015-05-04:
Hm, this is a common issue with the message headers not compiling before the executables. I have a version of the CMakelists.txt that forces the build of the messages first. A temporary way to get it to build is to use a smaller number of parallel threads while building the package. I'll see if I can push that commit sometime.

chrisspen, 2015-05-04:
Yeah, googling found me [this similar question](http://answers.ros.org/question/64448/trouble-with-catkin-compiling-node-and-generating-messages-in-same-package/), which mentions this bug in catkin. I worked around it by running catkin twice like:

    catkin_make -j1
    catkin_make

which feels a little hacky.

Yes, please push your fixed CMakeLists.txt when you're able.

oblivione, 2016-08-23:
cd lsd_slam/

git checkout catkin 
you run " ls"

response should :

LICENSE lsd_slam lsd_slam_core lsd_slam_viewer README.md

Then add 
"build_depend>cmake_modules</build_depend"

"run_depend>cmake_modules</run_depend"

in "package.xml" both in the folder "lsd_slam/lsd_slam_viewer" and in the folder "lsd_slam/lsd_slam_core".

In "CMakeFiles.txt" both in the folder "lsd_slam/lsd_slam_viewer" and in the folder "lsd_slam/lsd_slam_core"

add:
find_package(cmake_modules REQUIRED)

check the line below (X11 may be mising ) 
target_link_libraries(lsdslam ${FABMAP_LIB} ${G2O_LIBRARIES} ${catkin_LIBRARIES} csparse cxsparse X11)

Then u can build your workspace by 
cd path_to_catkin_ws
catkin_make

Then again if got some error of header file missing then 
"CMakeFiles.txt" in the folder "lsd_slam/lsd_slam_core" add:
add_dependencies(lsdslam lsd_slam_viewer_generate_messages_cpp)
add_dependencies(live_slam lsd_slam_viewer_generate_messages_cpp)
add_dependencies(dataset lsd_slam_viewer_generate_messages_cpp)

"CMakeFiles.txt" in the folder "lsd_slam/lsd_slam_viewer" add:
add_dependencies(viewer lsd_slam_viewer_generate_messages_cpp)

romdos, 2016-10-29:
I have the next error:
~/ros_workspace/catkin/src/lsd_slam/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:24:34: fatal error: g2o/core/base_vertex.h: No such file or directory
compilation terminated.

apesIITM, 2017-03-18:
@oblivione following the same steps, it gave this error [Eigen]:
=-=-=-=-=-=-=-=
`By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
asked CMake to find a package configuration file provided by "Eigen3", but
CMake did not find one.

Could not find a package configuration file provided by "Eigen3" with any
of the following names:

Eigen3Config.cmake
eigen3-config.cmake

Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
"Eigen3_DIR" to a directory containing one of the above files. If "Eigen3"
provides a separate development package or SDK, be sure it has been
installed.`
=-=-=-=-=-=-=-=

romdos, 2017-03-18:
You must set a path to Eigen library.

Toumi0812, 2017-07-06:
Hi @oblivione,
How to find  "path_to_catkin_ws" ?

Thanks