tum-vision/lsd_slam #80: rosmake error

antithing, 2015-03-18:
Hi! I am attempting to install lsd-slam on ubuntu 12.04, and ros fuerte.

I have cloned the git repository.

when i run 'rosmake lsd-slam'

it gives me:

...[ rosmake ] All 13 linessd_slam_core: 4.6 sec ]      [ 1 Active 14/15 Complete ]
{-------------------------------------------------------------------------------
  [  7%] Built target ROSBUILD_gencfg_cpp
  [  7%] Built target rospack_gencfg
  [  7%] Built target rospack_genmsg_libexe
  [  7%] Built target rosbuild_precompile
  [ 11%] Building CXX object CMakeFiles/lsdslam.dir/src/DataStructures/Frame.cpp.o
  /tmp/ccyyUgwk.s: Assembler messages:
  /tmp/ccyyUgwk.s:3188: Error: no such instruction: `vfmsub312ss 4(%rcx),%xmm1,%xmm7'
  /tmp/ccyyUgwk.s:8998: Error: no such instruction: `vfmadd312sd 0(%rbp),%xmm3,%xmm6'
  /tmp/ccyyUgwk.s:9002: Error: no such instruction: `vfmadd312sd 8(%rbp),%xmm3,%xmm5'
  /tmp/ccyyUgwk.s:9005: Error: no such instruction: `vfmadd312sd 16(%rbp),%xmm3,%xmm4'
  make[2]: *** [CMakeFiles/lsdslam.dir/src/DataStructures/Frame.cpp.o] Error 1
  make[1]: *** [CMakeFiles/lsdslam.dir/all] Error 2
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package lsd_slam_core written to:
[ rosmake ]    /home/b/.ros/rosmake/rosmake_output-20150318-154231/lsd_slam_core/build_output.log
[rosmake-1] Finished <<< lsd_slam_core [FAIL] [ 4.68 seconds ]                 
[ rosmake ] Halting due to failure in package lsd_slam_core.
[ rosmake ] Waiting for other threads to complete.
[ rosmake ] Results:                                                           
[ rosmake ] Built 15 packages with 1 failures.                                 
[ rosmake ] Summary output to directory             

Any help much appreciated!

xuetl, 2015-07-21:
same  issue. do you solve the problem?@antithing

antithing, 2015-07-21:
nope. Sadly I never got it to work. Good luck!

Dvad, 2015-07-29:
That happens in 12.04 with fairly new cpu and old gcc.
You can try to remove '-march=native' from the following line in CMakeLists.txt.

    set(CMAKE_CXX_FLAGS
       "${CMAKE_CXX_FLAGS} ${SSE_FLAGS} -march=native  -std=c++0x"
    )