tum-vision/lsd_slam #81: Moving Objects

colek42, 2015-03-20:
I am trying to localize moving objects.  Is is possible with LSD slam, given a window, to extract depth information for objects that are moving from frame to frame, i.e. a walking person?  I will be matching feature vectors in R^2 (HOG+SVM) and attempting to localize them in R^3.  I noticed that the moving objects turn black in the visulizer.  Much appreciated if you could point me in the right direction.

amiltonwong, 2015-03-20:
Hi, colek42,
Could you present why it has significance to localize the moving objects? Or you want to do a dynamic reconstruction of the moving objects?

colek42, 2015-03-20:
I do not need to do reconstruction of the moving object.  Right now our research is focused on tracking moving objects using low cost MAVs.  Use case will be avoidance/following.  Reconstruction using monocular slam is a difficult problem.

amiltonwong, 2015-03-20:
So for your scenario, object tracking approach may meet your needs

colek42, 2015-03-20:
Matching SURF/SIFT features on humans does not work well.  It looks like the depth information make be marked as invalid in somwhere in DepthEstimation/DepthMap.cpp.  

Everytime I get a new key frame I am going to search for matching feature vectors (humans) in R^2.  If I have a match I want to mark the 'box', and  and segment the points in R^3 that reside in that box, in a seperate data structure.  Right now it looks like those points get discarded.  I want to use them,  I have other aproaches as well if you don't think this will work well, we do want to use LSD SLAM for other reasons.  I apologize if this does not make sense, I am still at the hypothesis stage.