tum-vision/lsd_slam #85: Tracking lost while runing lsd_slam with rgb-d dataset

liqile, 2015-03-26:
I ran rgbd_dataset_freiburg1_360.bag and rgbd_dataset_freiburg2_pioneer_slam.bag.
when running rgbd_dataset_freiburg1_360.bag ,the lsd-slam lost tracking at frame 33
when running rgbd_dataset_freiburg2_pioneer_slam.bag the lsd-slam lost tracking at frame 798

I ran the lsd_slam_core with commands:
rosrun lsd_slam_core live_slam image:=/camera/rgb/image_color camera_info:=/camera/rgb/camera_info

amiltonwong, 2015-03-26:
@liqile,
I have the same test result as yours. It seems the reason in fast up and down moving in "freiburg1_360" and jittering movement in "freiburg2_pioneer_slam".

I have another question: According to following example generated from RGB-D camera (which has full dense depth map)
https://www.dropbox.com/s/xqxjf7bnokbekv7/01.jpg?dl=0

Can we also generate the approximate result using LSD-SLAM in semi-dense depth map manner?

liqile, 2015-03-26:
@amiltonwong ...
eh..I have no idea

liqile, 2015-03-26:
@amiltonwong I have run lsd-slam on another rgb-d dataset. but it also lost tracking...

liqile, 2015-03-30:
@amiltonwong ,I knew the reason now, we should write our own calibration file instead of using the ros defualt camera info in the bag

vins31, 2015-04-17:
It did not improve the results in my case. Did it for you?

liqile, 2015-04-17:
@vins31 
I haven't run lsd_slam with feiburg2_pioneer_slam..

liqile, 2015-04-24:
@vins31 n
I write the calibration file, I do not use the calibration parameters from the topic "/camera/rgb/camera_info" ,then the lsd-slam works well

vins31, 2015-04-24:
I used this calibration file for the fr1 scenes, which is quite close of the "/camera/rgb/camera_info"
```
517.3 516.5 318.6 255.3 0 0 0 0
640 480
full
640 480
```
Is it what you used?

liqile, 2015-04-24:
@vins31 
I used
517.3	 516.5	 318.6	 255.3	 0.2624	 -0.9531	 -0.0054	 0.0026	 1.1633

vins31, 2015-04-24:
I had already tested that.
The documentation specifies the calibration format is `fx fy cx cy k1 k2 p1 p2`, so the d4=1.1633 parameter is not interpreted in your case. This can be checked in the output of lsd_slam_core `In: 517.299988 516.500000 318.600006 255.300003 0.262400 -0.953100 -0.005400 0.002600`
With these values,  the images are completely distorted:
[image]

liqile, 2015-04-24:
@vins31 
hi,have you ever run lsd-slam with your own data?

vins31, 2015-04-24:
No, this freiburg1 360

lilouch, 2015-04-28:
I have also the same results. In general, LSD Slam works only well with their dataset.
I tried with the others dataset like KITTI and data provided by the computer vision group, and it's very bad.

Did it improve the results by setting manually the calibration parameters ?

liqile, 2015-04-28:
@lilouch what about running your own dataset?

lilouch, 2015-04-28:
I don't have mt own dataset for now because I wanted to try in existing dataset.

bjornph, 2015-06-22:
Have any of you gotten this to work? I am struggling with the same problems as @vins31