tum-vision/lsd_slam #91: Segmentation fault (core dumped) in ROS indigo + Ubuntu 14.04 64bit

Maverobot, 2015-04-16:
Hello guys,

I'm trying to use lsd_slam with my own rosbag file. However, whenever i run the lsd_slam_core, there is a "Segmentation fault". Here is the info printed in terminal.

WAITING for ROS camera calibration!
Received ROS Camera Calibration: fx: 551.589722, fy: 550.291321, cx: 308.271118, cy: 229.201431 @ 640x480
RECEIVED ROS camera calibration!
Started mapping thread!
Started  constraint search thread!
Started optimization thread 
Doing Random initialization!
Segmentation fault (core dumped)

I tracked down the error, and it happened in "Frame.cpp":

    for(float* pt = data.image[0]; pt < maxPt; pt++)
    {
        *pt = *image;
        image++;
    }

on the line of "  *pt = *image;  ". I tried to fix it but failed. Does someone have the same problem? I would really appreaciate some help or suggestions.

NoWiS-, 2015-06-11:
Hi,

I ran into the same problem with a NAO v5.

The default parameters of naoqi_sensors/camera.py node set the camera on QVGA (320x240) but advertise a VGA (640x480) resolution on the camera_info node (using nao_camera_bottom_640x480.yaml default file).

Maverobot, 2015-08-28:
@NoWiS-  Thanks. It seems i made the same mistake. :P

closed 2015-08-28 by Maverobot

TimingSpace, 2016-04-26:
@Maverobot I meet similar problem with you, could you please share how you solve the problem?