tum-vision/lsd_slam #92: `keyframeMsg.h` and `keyframeGraphMsg.h` are not included in the repository.

xocoatzin, 2015-04-16:
I've noticed that `keyframeMsg.h` and `keyframeGraphMsg.h` are referenced in the project `lsd_slam_visuaizer`, but they are not included in the source code. Am I missing something? 

I'm in `d1e6f0e1a027889985d2e6b4c0fe7a90b0c75067` in `master`

chrisspen, 2015-05-04:
I ran into this as well. My guess is it has something to do with the lsd_slam_viewer/msg/keyframeMsg.msg and lsd_slam_viewer/msg/keyframeGraphMsg.msg files. Maybe some part of the build process is supposed to convert these into .h files and is broken?

kilmarnock, 2015-06-11:
I ran into this as well. The files appear if you build lsd_slam_visualizer. Make sure to build it before lsd_slam_core.

xocoatzin, 2015-06-15:
Still can't see why these are just not simply included. 

I was building it without ROS, so had to recreate them manually (to the best of my understanding).  They can be found here >>> https://gist.github.com/142b8e6dc744172b5de4.git (modified with STL smart pointers, but it doesn't really mater)

freshNfunky, 2016-08-28:
thanks for the gist, but there is are references in the headder files to: "../IOWrapper/OpenCV/WrapperTypes.h" which does not exist in the repository.

jgranleese01, 2016-11-11:
Have any of you made anymore progress using LSD-SLAM without ROS? I can't get the point cloud to load without it apparently

ashish-nr, 2018-10-08:
I faced the same issue (with ROS Kinetic on an Ubuntu 16 machine).
Note that the headers are generated at build time in the `devel/include/lsd_slam_viewer` directory via catkin's message_generation functionality.

This worked for me:

- Copy the `msg` directory under `lsd_slam_viewer`, into `lsd_slam_core`. 
- Next we need to generate the two headers, one set each for lsd_slam_viewer and lsd_slam_core.
We do this for the viewer by adding an additional line in lsd_slam_viewer/CMakelists.txt: 
```add_dependencies(viewer lsd_slam_viewer_generate_messages_cpp)``` under the add_executable line for viewer.
As for lsd_slam_core/CMakelists.txt it's a little more work:
Add ```message_generation``` to the list of `find_package(catkin REQUIRED COMPONENTS)`
Then add the lines 
```add_message_files(DIRECTORY msg FILES keyframeMsg.msg keyframeGraphMsg.msg)```
```generate_messages(DEPENDENCIES)``` 
above the `generate_dynamic_reconfigure_options` line.
Then add the ```add_dependencies``` lines for the two executables, similar to the change made for `lsd_slam_viewer`, except in this case we specify ```lsd_slam_core_generate_messages```
- Lastly, replace all instances of `lsd_slam_viewer` in the `IOWrapper/ROS/ROSOutput3DWrapper.cpp` file with `lsd_slam_core`

Use my fork's [lsd_slam_viewer/cmakelists.txt](https://github.com/ashish-nr/lsd_slam/blob/catkin-mac_edit1/lsd_slam_viewer/CMakeLists.txt), [lsd_slam_core/cmakelists.txt](https://github.com/ashish-nr/lsd_slam/blob/catkin-mac_edit1/lsd_slam_core/CMakeLists.txt) and [ROSOutput3DWrapper.cpp](https://github.com/ashish-nr/lsd_slam/blob/catkin-mac_edit1/lsd_slam_core/src/IOWrapper/ROS/ROSOutput3DWrapper.cpp) for reference.
Hope this helps!