tum-vision/lsd_slam #93: LSD_SLAM missing dependency lib2go ros indigo

iamfarhan007, 2015-04-17:
Hi 
when i try to install dependencies for LSD_SLAM on ubunto 14.04.1 ROS indigo i get 

E: Unable to locate package ros-indigo-libg2o

Thanks

Farhan

Tgaaly, 2015-06-19:
make sure you do the step in the installation that installs the dependencies (depending on your ROS version):
sudo apt-get install ros-fuerte-libg2o liblapack-dev libblas-dev freeglut3-dev libqglviewer-qt4-dev libsuitesparse-dev libx11-dev

ramanpreet9, 2015-06-24:
This is the message I get for libqglviewer-qt4-dev :

E: Package 'libqglviewer-qt4-dev' has no installation candidate

The instructions on lsd_slam say libqglviewer-dev  which works fine for lsd_slam.

I am however trying to compile dvo_slam on indigo and facing this issue. How do i in some way redirect   libqglviewer-qt4-dev to libqglviewer-dev

jmcmahon443, 2016-03-25:
After fresh install of Ubuntu 14 and ROS Indigo, I have the same error.
E: Package 'libqglviewer-qt4-dev' has no installation candidate

ramanpreet9, 2016-06-11:
same issue.. any solutions to this?