tum-vision/lsd_slam #95: Problem during the installation

lilouch, 2015-04-20:
Hello,

I try to install lsd-slam but I encountered some problem. I did the command line rosmake lsd_slam but I got errors during the ocmpilation:
CMakeFiles/viewer.dir/src/PointCloudViewer.cpp.o: In function `PointCloudViewer::PointCloudViewer()':
  PointCloudViewer.cpp:(.text+0x1776): undefined reference to `QGLViewer::QGLViewer(QWidget*, QGLWidget const*, QFlags<Qt::WindowType>)'
  CMakeFiles/viewer.dir/src/PointCloudViewer.cpp.o:(.rodata._ZTV16PointCloudViewer[vtable for PointCloudViewer]+0x18): undefined reference to `QGLViewer::qt_metacast(char const*)'
  CMakeFiles/viewer.dir/src/PointCloudViewer.cpp.o:(.rodata._ZTV16PointCloudViewer[vtable for PointCloudViewer]+0x20): undefined reference to `QGLViewer::qt_metacall(QMetaObject::Call, int, void**)'
  CMakeFiles/viewer.dir/src/PointCloudViewer.cpp.o:(.rodata._ZTV16PointCloudViewer[vtable for PointCloudViewer]+0x310): undefined reference to `QGLViewer::setPlayPathKeyboardModifiers(QFlags<Qt::KeyboardModifier>)'
  CMakeFiles/viewer.dir/src/PointCloudViewer.cpp.o:(.rodata._ZTV16PointCloudViewer[vtable for PointCloudViewer]+0x318): undefined reference to `QGLViewer::setAddKeyFrameKeyboardModifiers(QFlags<Qt::KeyboardModifier>)'
  collect2: ld a retourné 1 code d'état d'exécution
  make[3]: *** [../bin/viewer] Erreur 1
  make[3]: quittant le répertoire « /home/samoun/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build »
  make[2]: *** [CMakeFiles/viewer.dir/all] Erreur 2
  make[2]: quittant le répertoire « /home/samoun/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build »
  make[1]: *** [all] Erreur 2
  make[1]: quittant le répertoire « /home/samoun/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build »
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package lsd_slam_viewer written to:
[ rosmake ]    /home/sl001093/.ros/rosmake/rosmake_output-20150420-114707/lsd_slam_viewer/build_output.log
[rosmake-2] Finished <<< lsd_slam_viewer [FAIL] [ 15.83 seconds ]                                                      
[ rosmake ] Halting due to failure in package lsd_slam_viewer. 
[ rosmake ] Waiting for other threads to complete.     
[ rosmake ] Results:                                                                                                   
[ rosmake ] Built 41 packages with 1 failures.                                                                         
[ rosmake ] Summary output to directory                          

I Installed all the dependencies.

I use ROS Hydro + Ubuntu 12.04

rambleramble, 2015-07-03:
same issue,  ROS Hydro+Ubuntu 12.04

xuetl, 2015-07-21:
same issue,ROS fuerte+Ubuntu 12.04

bl4ckc00k1e, 2015-08-04:
Same issue, ROS Indigo+Ubuntu 14.04, Out put in Verbose:

[rosmake-2] Starting >>> lsd_slam_viewer [ make ]                                                                                                                                               
[ rosmake ] Last 40 linesd_slam_viewer: 2.4 sec ]                                                                                                                    [ 1 Active 40/42 Complete ]
{-------------------------------------------------------------------------------
  main_stitchVideos.cpp:(.text+0x8f01): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0x8f30): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0x8fc9): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0x8fdb): referencia a `cv::String::deallocate()' sin definir
  CMakeFiles/videoStitch.dir/src/main_stitchVideos.cpp.o: En la función `inlayVid(int, char**)':
  main_stitchVideos.cpp:(.text+0x9c62): referencia a `cv::imread(cv::String const&, int)' sin definir
  main_stitchVideos.cpp:(.text+0x9c6a): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0x9cb3): referencia a `cv::imread(cv::String const&, int)' sin definir
  main_stitchVideos.cpp:(.text+0x9cbd): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0x9d02): referencia a `cv::imread(cv::String const&, int)' sin definir
  main_stitchVideos.cpp:(.text+0x9d0c): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0x9d5a): referencia a `cv::imread(cv::String const&, int)' sin definir
  main_stitchVideos.cpp:(.text+0x9d67): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0x9db1): referencia a `cv::imread(cv::String const&, int)' sin definir
  main_stitchVideos.cpp:(.text+0x9dbe): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0xa629): referencia a `cv::String::allocate(unsigned long)' sin definir
  main_stitchVideos.cpp:(.text+0xa654): referencia a `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)' sin definir
  main_stitchVideos.cpp:(.text+0xa661): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0xa6c6): referencia a `cv::String::allocate(unsigned long)' sin definir
  main_stitchVideos.cpp:(.text+0xa6e7): referencia a `cv::imshow(cv::String const&, cv::_InputArray const&)' sin definir
  main_stitchVideos.cpp:(.text+0xa6f4): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0xaec5): referencia a `cv::String::allocate(unsigned long)' sin definir
  main_stitchVideos.cpp:(.text+0xaee8): referencia a `cv::String::allocate(unsigned long)' sin definir
  main_stitchVideos.cpp:(.text+0xaf08): referencia a `cv::String::allocate(unsigned long)' sin definir
  main_stitchVideos.cpp:(.text+0xaf28): referencia a `cv::String::allocate(unsigned long)' sin definir
  main_stitchVideos.cpp:(.text+0xaf46): referencia a `cv::String::allocate(unsigned long)' sin definir
  main_stitchVideos.cpp:(.text+0xb146): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0xb1e2): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0xb20e): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0xb23f): referencia a `cv::String::deallocate()' sin definir
  main_stitchVideos.cpp:(.text+0xb270): referencia a `cv::String::deallocate()' sin definir
  CMakeFiles/videoStitch.dir/src/main_stitchVideos.cpp.o:main_stitchVideos.cpp:(.text+0xb33e): más referencias a `cv::String::deallocate()' sin definir a continuación
  collect2: error: ld returned 1 exit status
  make[3]: *** [../bin/videoStitch] Error 1
  make[3]: se sale del directorio «/opt/ros/rosbuild_ws/lsd_slam/lsd_slam_viewer/build»
  make[2]: *** [CMakeFiles/videoStitch.dir/all] Error 2
  make[2]: se sale del directorio «/opt/ros/rosbuild_ws/lsd_slam/lsd_slam_viewer/build»
  make[1]: *** [all] Error 2
  make[1]: se sale del directorio «/opt/ros/rosbuild_ws/lsd_slam/lsd_slam_viewer/build»
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package lsd_slam_viewer written to:
[ rosmake ]    /home/bl4ckc00k1e/.ros/rosmake/rosmake_output-20150804-193623/lsd_slam_viewer/build_output.log                 
[rosmake-2] Finished <<< lsd_slam_viewer [FAIL] [ 2.41 seconds ]                                                                                                                                
[ rosmake ] Halting due to failure in package lsd_slam_viewer. 
[ rosmake ] Waiting for other threads to complete.                                                                              
[rosmake-7] [ Build Terminated Thread Exiting ]                                                                                                                                                 
[rosmake-3] [ Build Terminated Thread Exiting ]                                                                                                                                                 
[rosmake-5] [ Build Terminated Thread Exiting ]                                                                                                                                                 
[... cut ...]
[ rosmake ] Results:                                                                                                                                                                            
[ rosmake ] Built 41 packages with 1 failures.                                                                                                                                                  
[ rosmake ] Summary output to directory                                                                                                                                                         
[ rosmake ] /home/bl4ckc00k1e/.ros/rosmake/rosmake_output-20150804-193623

Geaper, 2016-05-24:
same problem. Any fix?

bchu, 2016-07-06:
I had errors similar to this when I had multiple versions of libQGLViewer installed, which I fixed by removing all traces I could find (`find /usr/ -name '*QGLView*"`)  and reinstalling.

zhuangxiaopi, 2019-06-23:
fixed?I encouter same problem .....

mirellameelo, 2020-07-20:
I'm using a [version](https://github.com/x1597275/LSD-SLAM-with-ubuntu1804-and-melodic) for ROS melodic, but I had the same problem... I used the following commands:

```hl
sudo apt install libqglviewer-dev-qt4  
cd /usr/lib/x86_64-linux-gnu  
sudo ln -s libQGLViewer-qt4.so libQGLViewer.so
```

It may occurs more errors. [Here](https://blog.csdn.net/softimite_zifeng/article/details/78632653) some help for next issues.