tum-vision/lsd_slam #100: Running lsd_slam_viewer on ODROID-XU3

roosemberth, 2015-04-27:
Hello,

I'm trying to run LSD Slam on ODROID-XU3, I'm able to correctly initialize lsd_slam_core but when I run lsd_slam_viewer I get tons of errors:

SCP-Svarog@SCP-Svarog:/Development/lsd_slam_datasets/LSD_room/images$ rosrun lsd_slam_viewer viewer                                                                                                                                                                      [46/46]
Started QApplication thread
QEglContext::createSurface(): Unable to create EGL surface, error = 0x3003
QGLContext::makeCurrent(): Cannot make invalid context current
QGLContext::makeCurrent(): Cannot make invalid context current
Started ROS thread
QGLContext::makeCurrent(): Cannot make invalid context current
QEglContext::swapBuffers(): "Bad surface (0x300D)" 
[ INFO] [1430115665.790735658]: lsd_slam_viewer started
QGLContext::makeCurrent(): Cannot make invalid context current
QEglContext::swapBuffers(): "Bad surface (0x300D)" 
QGLContext::makeCurrent(): Cannot make invalid context current

Can this be due to OpenGL 3.0 ES instead of OpenGL?

Best Regards!
Roosemberth Palacios

roosemberth, 2015-06-05:
Hello?, any ideas?

antoan, 2015-06-22:
The viewer requires the odroid GPU drivers to be setup since it renders with OpenGL. The odroid forum is a good place to start.