tum-vision/lsd_slam #101: LSD SLAM with KITTI dataset

lilouch, 2015-04-27:
Hello,

I would like to know if the algorithm works well with KITTI Dataset because I ran the algorithm for 11 sequences and it didn't work for all. either it crashed or either the results is really really bad.

So i wonder if it's because of the fps required by the algorithm (minimum 30) ? Or because of some parameters ?

Thank you

amiltonwong, 2015-04-27:
I also have the similar result.

One issue is that the capture rate is 10fps in KITTI dataset, which is far away to the requirement of LSD-slam,
Second, there are many self-rotating movement like turning a corner. The tracker loses tracking easily.

lilouch, 2015-04-28:
Thank for your answer.

So no way to make it work ? Except if I have data at 40-50 fps ?
And LSD-SLAM not designed for urban environment so, right ?

Do you know why self-rotating movement is not so good ?

thank you.

Thyrf, 2015-06-02:
Hello,

I would suggest that LSD-SLAM is not working as the displacement between every frame is too large. If you read the paper Semi-Dense Visual Odometry for a Monocular Camera on which the algorithm is build on you will be able to see in Figure 3 how a large baseline involves lots of local minima. While the filtering approach of the depth map and the weighted least square formulation of the dense tracking help robustify the system, it might reach its robustness limits with this dataset and converge to something wrong.
But to my opinion this has nothing to do with the fact that KITTI is a urban environment. I remember seeing video where it worked pretty well in such environments.
Cheers

JakobEngel, 2015-06-02:
Correct, the two reasons why KITTI doesnt work are
1) low frame-rate, hence very fast inter-frame motion (especially in highway-scenes, like sequence 01).
2) strong rotations without much translation during turns
3) relatively small field of view, and forward-motion.

(1) can be compensated by using a somewhat meaningful motion model (e.g. assuming constant motion) to initialize tracking.
(2) and (3) can be solved by using stereo, instead of mono ;)

There might be a stereo-version of LSD-SLAM (which works well on e.g. KITTI) some time soon - however as of now it's undecided whether it's going to be open-source.

pmolina7, 2015-12-18:
The stereo LSD SLAM results on the paper look great. It would be great if it was open source so it can be tried out on other datasets. Thanks,

Thyrf, 2015-12-19:
Hello,

are you referring to the paper I submitted to ICRA this year?
If that is the case I'm glad you like the results. I was also pretty
pleased with them too. It felt like a very intuitive and simple way to deal
with the problem and worked very well. I have to admit I felt a bit gutted
at the time I did it. I explain myself a bit: I programmed that in early
July 2014,  I felt like I had nice contributions to offer to the SLAM
community and then saw your LSD SLAM  ECCV publication with most of the
contributions I wanted to propose included in it. It was hard but at least
I really enjoyed the work of your team. The paper is nicely written and the
results so nice.
Anyway this made me feel like I had not enough things to propose for a
publication and I left the project on the side. Only later this year as I
didn't see any similar approach being proposed for large movements I
thought I'd work on written a clean code, making it open source and try to
publish it. So if that was only up to me I'd open source it. I already
asked D. Murray his approval but didn't have it yet. I'll get back to him
and let you know.
Cheers

caap95, 2017-10-14:
How can i check my fps during execution?

ashwinjoseph95, 2024-07-31:
> The stereo LSD SLAM results on the paper look great. It would be great if it was open source so it can be tried out on other datasets. Thanks,

Hi just wanted to know if the LSD-SLAM Stereo version code is available on git somewhere?