tum-vision/lsd_slam #102: live_slam is running really slow?

tarazena, 2015-04-27:
I have MacBook Pro 2013 model, Installed Ubuntu 14.04 and ROS Inidigo. 
Whenever i try to run live_slam on my webcam, the frame rate for analyzing images is pretty low (3-1 FPS), Is that normal?
When i try to use Dataset, it works really fast, only the live_slam is slow.

Also can you please update the command for the dataset slam? im using dataset only not dataset_slam.

Thanks

bconvens, 2015-04-28:
I have the same problem. I also use ubuntu 14.04 ROS indigo and I run live_slam on my AR drone 2.0 video stream, but the framerates are very very low. Sometimes the frames hang for more than 3s. And most of the time, the program stops running if you rotate the drone (even slow rotations).
First I thought it was my camera calibration file that was not good, but I used the same values as in the ardrone_autonomy camera calibration file. Also a real straight line looks straight on the screen, so I assume that this isn't the problem or am I wrong?
Do you think it is maybe graphically to heavy to display the depth map? 
How did you achieve in running this almost real time? Any proposals?

In my project I want to use the values of the relative depth (where can I find them? Filename, line?) to test an object avoidance algorithm. Maybe this helps you to see which part of your code I need as input for my project.

Thank you very much for the help!

Tgaaly, 2015-06-21:
I have the same problem with my ardrone!! I was able to disable the depth display window for faster mapping. I only view the map from the lsd viewer.

My problem is that the map isn't very good at all. Any ideas?

mihaibujanca, 2015-10-19:
+1
It works extremely slow for me. I have an Intel i7 quad core 2.6GHz, so the processing power is not the issue, I think. I'm getting <1Hz most of the time, sometimes it goes to 1.5Hz, but that's it.

gblikas, 2016-08-11:
Same problem, even in 2016...

caap95, 2017-10-14:
How can i check what fps am i getting during the execution?

paul3333, 2021-02-27:
Has anyone fixed that error? I'm facinig the exact same issues on my ardrone.