tum-vision/lsd_slam #105: EGL suggested using Visual ID

DuskoMurtovski, 2015-05-06:
Hi,

I am running lsd_slam on Odroid U3 and until now it was working just fine. Bun (unfortunately) the system was updated from ubuntu 14.02 to 14.04 and now i am getting this warning when i try to start the viewer:

Warning: EGL suggested using X Visual ID 240 (ARGB0888) for EGL config 9 (ARGB8888), but this is incompatible

After this the thread exits and the viewer is not started. I suspect that the problem is either in some driver or in some opengl dependency.

I am not sure what i should do.

Please help

mtyong55555, 2015-07-15:
Hi there:
I've met the same problem when I was running qt program on odroid u3, have you solved it yet?
Thanks.

DuskoMurtovski, 2015-07-16:
Hi, i haven't solved the problem. I quit using the program on odroid. The problem seems to be the version of the OpenGL driver which is 2.1 in ubuntu 14.04.2 and odroid supports 2.0.
I might be completely wrong, i haven't tested the program since my last post. If i found something i will post it.

mtyong55555, 2016-01-27:
Hi:
I think I've solved this problem, I just changed my qt complier from qt4 to qt5, and this problem was gone.
I don't know what qt has improved from qt4 to qt5, but it really solved this problem and some others.

DuskoMurtovski, 2016-01-28:
That is great news, i unfortunately destroyed my odroid u3 :), (working with raspberry pi 2 now) but i will keep this in mind.