tum-vision/lsd_slam #109: running slowly without ros

liqile, 2015-05-21:
hi, @JakobEngel a
if I run ros with ros, it is very fast, 7~11hz
if I run ros without ros, it is very slow, less than 1hz,
can you give me some advices while running without ros?
thanks a lot

kryptoneit, 2015-07-03:
Hi @liqile 
Im trying to run lsd-slam without ROS but Im not able to make out how it can be done.
Can you give me some tips or suggestions to make it to run without ROS ?
Thanks

mkassner, 2015-07-03:
This has to do with the external libs used in the no-ros branches.
Just build with ros and use the dylib without ros componets.
This way lsd-slam will use ros g2o etc but all the IO is connected at
runtime with the tool of your choice.

liqile, 2015-07-07:
@kryptoneit I think you can change some code to get rid of the ros lib. and also change the cmakelist file, can you tell me what's the problem?

yuanshucong, 2015-12-18:
@liqile hi, i am trying to understand the code,but i find it read hard to do it without "eclipse step by step debug", but when i cant import the project into eclipse,there are too many bugs,can you tell me how to run the code with the eclipse build???