tum-vision/lsd_slam #110: lsd_slam build for android

aivijay, 2015-05-25:
Has anyone been able to build lsd_slam for android? I had been trying to build with https://github.com/williammc/lsd_slam which seems to have some updates specific for android build. But there are a bunch of arm specific build dependencies of various libraries which is needed for the build. I had been able to build a linux build without much problems. But going through the cmake script specific for android, seems to revolve around local references to build of libraries for arm and used within to do the android build. I dont know whether there were successful builds with this code base. It would be great to know that. Also how was g2o etc build for arm or am I mistaken here. Seem from 

lsd_slam/cmake/LsdSlamDependencies_Config.cmake.in

All the references are to local prebuilts for arm. Was able to get boost, freeglut, opencv builds for arm. But any ideas on g2o and opengl seems to be required for these builds. How was it build for this fork of lsd_slam as well?

It will be great if someone can respond with their experience or any pointers how it was build or whether it can be built.

Thanks,
-vijay

aivijay, 2015-05-28:
I head from William that the code base is build able for android. Will see if I can do my own build.

closed 2015-05-28 by aivijay

amiltonwong, 2015-05-29:
Hi, aivijay,
Is the code you mentioned from "https://github.com/williammc/lsd_slam" ?

dpproduction, 2017-07-28:
Does anybody compile it for Android?