tum-vision/lsd_slam #115: LSD-Slam capture does not start

holzkohlengrill, 2015-06-03:
When I am trying to start LSD-Slam like:

rosrun lsd_slam_core live_slam image:=/image_raw _calib:='<path_to_my_calibrationfile.cfg' 

I get the following output (no window appears, ...):

Reading Calibration from file /home/marcel/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/calib/OpenCV_example_calib.cfg ... found!
found OpenCV camera model, building rectifier.
Input resolution: 752 480
In: 500.000000 500.000000 350.000000 240.000000 0.000000 0.000000 0.000000 0.000000
Out: Crop
Output resolution: 640 480
Started mapping thread!
Started  constraint search thread!
Started optimization thread 

Another point (which could be related to this issue) is that when I view a pointcloud dataset I got colour information (just the viewer is shown).

Thanks.

kilmarnock, 2015-08-12:
The window appears as soon as you provide images on the right topic

xiaokeme, 2015-08-14:
@kilmarnock ,I encountered the same question as @holzkohlengrill ,but how to provide images on the right topic?

xiaokeme, 2015-08-14:
the steps I did is shown: https://github.com/tum-vision/lsd_slam/issues/141

kilmarnock, 2015-08-14:
@xiaokeme , try to set up your camera without using lsd_slam. Verify your setup:  
rostopic hz /image_raw 
should give you the current frames per second on this topic. If you do not get any image, try to ask your question in ros-answers, it is not lsd_slam related then but rather a problem of your camera setup.

xiaokeme, 2015-08-14:
@kilmarnock ,thank you very much!

mihaibujanca, 2015-10-14:
@kilmarnock I'm having the same issue.
The topic I am using is clearly published, since I can view it using image_view and I can do 

rostopic hz /image_raw

For me it only gets as far as 
```
found ATAN camera model, building rectifier.
Input resolution: 1920 1080
In: 0.527334 0.827306 0.473568 0.499436 0.000000
NO RECTIFICATION
Output resolution: 752 480
Prepped Warp matrices
Started mapping thread!
Started  constraint search thread!
Started optimization thread 
```

kilmarnock, 2015-10-15:
@mihaibujanca 
 lsd_slam waits for its images at the "image" topic. You can remap it:
 <remap from="image" to="/image_raw" /> or provide your image at /image and check it with 

rostopic hz /image

I corrected my post above, sorry for the mistake.

mihaibujanca, 2015-10-15:
@kilmarnock I am providing lsd slam with the image topic when I run it:

```
rosrun lsd_slam_core live_slam image:=/image_raw _calib:=mycalib.cfg
```

So I don't think that's the problem

kilmarnock, 2015-10-16:
what is the output of 
` rostopic info /image_raw`?

mihaibujanca, 2015-10-16:
@kilmarnock 

```
rostopic info /image_raw
Type: sensor_msgs/Image

Publishers:
* /video2ros (http://mihai-Aspire-E1-571G:41184)

Subscribers:

* /image_view_1444123124434232 (http://mihai-Aspire-E1-571G:60608)

```

Video2ros (package rbx1/rbx1_vision) is just publishing a video I recorded with a pi camera - at some point I'll try to stream the video from a pi and have it processed on my laptop. Looks like lsd_slam is not subscribing to it, though image_view has no problem with it, the image is displayed properly

kilmarnock, 2015-10-16:
Your assumption is right: lsd_slam is not subscribing. Try to use a launch file and remap the image topic in there:

````
 <node pkg="lsd_slam_core" type="live_slam" name="lsd_slam" output="screen">
        <remap from="image" to="/image_raw"/>
        <remap from="camera_info" to="/mycam/camera_info"/>
        <param name="calib" value="/path/to/config.cfg"/>
  </node>
````
You have to end providing images like so:
```` 
rostopic info /image_raw
Type: sensor_msgs/Image

Publishers: 
 * /somenode (http://somehost:56671/)

Subscribers: 
 * /lsd_slam (http://somehost:57934/)
 * /rviz_1444924448230228527 (http://otherhost:55785/)
````

mihaibujanca, 2015-10-16:
@kilmarnock 
Thanks, indeed remapping instead of passing it as a param worked.

All I have to do now is get a good calibration file, since I'm getting 
`TRACKING LOST for frame 29 (0.09% good Points, which is 41.50% of available points, DIVERGED)!`
with all videos.

Thanks a lot .