tum-vision/lsd_slam #116: Error on rosmake lsd_slam

nagneeve, 2015-06-04:
[ rosmake ] Last 40 linesd_slam_viewer: 24.2 sec ]                       [ 1 Active 40/42 Complete ]
{-------------------------------------------------------------------------------
  main_stitchVideos.cpp:(.text+0x9db1): undefined reference to `cv::imread(cv::String const&, int)'
  main_stitchVideos.cpp:(.text+0x9dbe): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0xa629): undefined reference to `cv::String::allocate(unsigned long)'
  main_stitchVideos.cpp:(.text+0xa654): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
  main_stitchVideos.cpp:(.text+0xa661): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0xa6c6): undefined reference to `cv::String::allocate(unsigned long)'
  main_stitchVideos.cpp:(.text+0xa6e7): undefined reference to `cv::imshow(cv::String const&, cv::_InputArray const&)'
  main_stitchVideos.cpp:(.text+0xa6f4): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0xaec5): undefined reference to `cv::String::allocate(unsigned long)'
  main_stitchVideos.cpp:(.text+0xaee8): undefined reference to `cv::String::allocate(unsigned long)'
  main_stitchVideos.cpp:(.text+0xaf08): undefined reference to `cv::String::allocate(unsigned long)'
  main_stitchVideos.cpp:(.text+0xaf28): undefined reference to `cv::String::allocate(unsigned long)'
  main_stitchVideos.cpp:(.text+0xaf46): undefined reference to `cv::String::allocate(unsigned long)'
  main_stitchVideos.cpp:(.text+0xb146): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0xb1e2): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0xb20e): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0xb23f): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0xb270): undefined reference to `cv::String::deallocate()'
  CMakeFiles/videoStitch.dir/src/main_stitchVideos.cpp.o:main_stitchVideos.cpp:(.text+0xb33e): more undefined references to `cv::String::deallocate()' follow
  collect2: error: ld returned 1 exit status
  make[3]: *** [../bin/videoStitch] Error 1
  make[3]: Leaving directory `/home/jae/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
  make[2]: *** [CMakeFiles/videoStitch.dir/all] Error 2
  make[2]: *** Waiting for unfinished jobs....

I am getting this error message and I am confused on what is going on. Should I resolve this error to do anything for LSD_slam? How can I get rid of this error? Please help me on this.
[ rosmake ] Built 41 packages with 1 failures. Does this tell me that something is built or the entire thing hasn't been built because of the error?

nagneeve, 2015-06-09:
Just in case anyone is wondering about this problem,,,

in the <b>CMakeLists.txt</b> file in your <b>lsd_slam_core folder</b>,

Add the following codes:

find_package(OpenCV REQUIRED)

Modify the following lines like pointed by arrow:

target_link_libraries(live_slam lsdslam ${catkin_LIBRARIES} ${G2O_LIBRARIES})
------->
target_link_libraries(live_slam lsdslam ${catkin_LIBRARIES} ${G2O_LIBRARIES} <b>${OpenCV_LIBS}</b>)

target_link_libraries(dataset lsdslam ${catkin_LIBRARIES} ${G2O_LIBRARIES})
-------->
target_link_libraries(dataset lsdslam ${catkin_LIBRARIES} ${G2O_LIBRARIES} <b>${OpenCV_LIBS}</b>)

closed 2015-06-09 by nagneeve

cipri-tom, 2016-02-01:
Hi!
Thank you for taking the time to write the solution you've found.
Unfortunately, this is not working for me. I suspect I have OpenCV 3 installed, which moved/removed these functions. Do you know how I could check which OpenCV it is trying to link with ?

cipri-tom, 2016-02-01:
Ok, so it needs OpenCV 3, but it is looking for OpenCV2. I solved it with the ugly hack in #163

gabrielarpino, 2016-06-23:
THANKS IT WORKED :)