tum-vision/lsd_slam #118: LSD_SLAM not compiling on oDroid XU3 with error of __m128 object not found

mohsaad, 2015-06-17:
Hello,

I've been trying to compile LSD_SLAM on the XU3 so I can put it on a quadcopter. However, whenever I try compiling the program, I always get the error:

__m128 not declared in this scope

I know that this has to do with the SSE instructions, but how can I compile with the SSE enabled? I'd like to have as much speedup as possible as to not waste computational resources.

Wramos, 2015-06-18:
Hi mohsaad, I came onto this same issue. I resolved it by going to Frame.cpp, and changing the if defined statements. More specifically it was the following block:

    #if defined(ENABLE_SSE)
        //first block of code
    #elif defined(ENABLE_NEON)
        //second block of code

I merely switched these two to stop seeing this error. It looks something like this:

    #elif defined(ENABLE_NEON)
        //second block of code
    #elif defined(ENABLE_SSE)
        //first block of code

And if I recall correctly, I also go into CmakeLists.txt and change the CMAKE_CXX_FLAGS from:

    set(CMAKE_CXX_FLAGS
        "${CMAKE_CXX_FLAGS} ${SSE_FLAGS} -march=native -std=c++0x"

to

    set(CMAKE_CXX_FLAGS
        "${CMAKE_CXX_FLAGS} -mfpu=neon -std=c++0x"

I don't think this should vary. This error is coming from the fact that arm processors (such as the one found in cellphones) do not support SSE natively like x86 and x64 architectures, so you must explicitly state that you need to use NEON. Don't take my word for it though, I'm just a student :stuck_out_tongue_closed_eyes: If someone else is able to correct me then go ahead.

mohsaad, 2015-06-18:
Hi Wramos,

That did the trick, thanks.

I also found that getting rid of
`add_definition("-DENABLE_SSE")`
and replacing it with 
`add_definition("-DENABLE_NEON")`

also works. You still need to add the `-mfpu=neon` option though, which is what I didn't know.

Thanks so much though!

closed 2015-06-18 by mohsaad