tum-vision/lsd_slam #123: run time error using dataset_slam

holazollil, 2015-06-26:
I install ROS and download lsd_slam code. I start ROS using roscore command on terminal. I have a folder,named grayImages, which contains list of images taken from an android tablet camera. I run " rosrun lsd_slam_core dataset_slam _files:=/....../grayImages _hz:=0 /...../opencvCalibFile.cfg ". I use calibration file under lsd_slam/lsd_slam_core/calib/OpenCV_example_calib.cfg.

I get below error : 

image grayImages/I1_000359.png has wrong dimensions - expecting 640 x 480, found 640 x 480. Skipping.

rambleramble, 2015-07-08:
I have the same issues here. I have a group of images, extracted from a video sequence.

holazollil, 2015-07-08:
You should change image width and height in your calib file.