tum-vision/lsd_slam #124: how to change the value of the paremeter KFUsageWeight and KFDsageWeight?

TomE8, 2015-06-26:
hi,
I want to change the values of the paremeters KFUsageWeight and the paremeter  KFDistWeight,  in order to improve tracking (as suggested). How should I do that? what files do I have to change?
Thanks...

bjornph, 2015-06-27:
I think you can change them by using rqt_reconfigure, try:
rosrun rqt_reconfigure rqt_reconfigure

http://wiki.ros.org/rqt_reconfigure

TomE8, 2015-06-27:
hi,
I have tried to do so and it didn't work- when I typed the command you have suggested: 
"rosrun rqt_reconfigure rqt_reconfigure",
it opens a screen (see attached image) which I can use in order to change part of the variables values, however the variables KFUsageWeight and KFDistWeight don't appear on this screen...  
I think I have to change some values in the program's files... The problem is that I dont know which file I should change...
any ideas?

[image]

bjornph, 2015-06-27:
You have to  select LSD_SLAM on the left hand side. You have only selected the viewer. 
[image]

TomE8, 2015-06-27:
hi,
I have checked, and my only option is to open the "viewer" tab, I dont have the "slam" tab you have marked in red...
I can see from the image you have sent, that you use only one terminal. In order to run the example file, I use 3 terminals:
one with- roscore
secound with- rosrun lsd_slam_viewer viewer
third with- rosrun lsd_slam_core dataset_slam _files:=<files> _hz:=<hz> _calib:=<calibration_file> (when I change the names of the files)
what are the commandes you type in the terminal?
thanks,
Tom

bjornph, 2015-06-28:
I do  run the same as you. Roscore, lsd_slam_viewer and lsd_slam_core. It works both when I run  "rosrun lsd_slam_core dataset_slam ......" and  "rosrun lsd_slam_core live_slam ......". So sorry, but then I don't know.

TomE8, 2015-06-29:
hi.
I did run it after all the commandes (and it doesn't work)... 

My main problem with the LSD slam program is that it losses tracking for images that I have taken with my camera. I have calibreted it, however something isn't working. I read in the LSD-git page that changing the variables KFUsageWeight and KFDsageWeight might help...

I will try to solve the problem by doing the calibration again:  I am using the matlab camera calibration app in order to get the calibration parameters. I know how to get the pareneters: fx, fy, cx, cy  from my calibration data however I dont understand what is the parameter 'd' (see matlab output in the image)?

Thank you for your help,
Tom
[image]

bjornph, 2015-06-29:
I'm sorry that did not work for you. It might be that you can change the default value in the file /lsd_slam/lsd_slam_core/cfg/LSDParams.cfg. This will probably not need a recompiling in order to apply the changes, but I'm not 100% sure.

I think the 'd' parameter is a lens distortion parameter. LSD-SLAM accepts different types of camera models (see https://github.com/tum-vision/lsd_slam#313-camera-calibration). If you use the FOV camera model then I would suggest calibrating your camera with the same type of calibration tool that they used (which is PTAM camera calibrator http://wiki.ros.org/ethzasl_ptam/Tutorials/camera_calibration) 

See this link for some more information about lens distortion: http://docs.opencv.org/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html