tum-vision/lsd_slam #126: Inverted Map

Annyv2, 2015-07-17:
Hi! I have been testing the LSD_SLAM On an Ardrone 2.0, and I had to lower the resolution for it to work fast enough so it could keep tracking, but i'm obaining an inverted map as referenced here #122 , it can be seen in the picture. What's causing this and how can I make it work correctly? Thanks in advance!
[image] #

v0n0, 2015-10-14:
Did you solve it?

Annyv2, 2015-10-15:
No, @v0n0. I ended up testing with the image like that since it didn't affect a lot what I needed this for. Sorry I can't help you further. If you find a solution let me know :D