tum-vision/lsd_slam #127: Relocalization

NilsL3, 2015-07-28:
Hello,

In the tum ardrone package, it is possible to occlude the camera with the hand and the algorithm has the capability of recovering from the loss of tracking, which was caused by the occlusion. Is this feature also present in lsd slam? If not, can relocalization be incorporated to lsd slam and if so, how?

I would appreciate your help. Thank you.

bjornph, 2015-07-28:
Relocalization is a part of LSD-SLAM, yes. It was introduced after the release of the paper.

NilsL3, 2015-07-28:
Thanks a lot for the help. I appreciate it a lot.

NilsL3, 2015-07-28:
I observed that with very radical movements of the camera, the tracking gets lost and sometimes when returning the camera to a prior position, which had been mapped, relocalization is successful and lsd slam is back to tracking. However, this is not always the case. In some situations when tracking is lost, I return the camera to a mapped position, but the tracking will not resume and relocalization is unsuccessful. Is there an explanation for this?

bjornph, 2015-07-29:
Relocalization is not 100% accurate. In short what a relocalization module would do is when you lose track, it will compare the current frame with your keyframes in order to see if it sees the same scene. If it is decided that it does, it can relocalize itself.

NilsL3, 2015-07-31:
Thanks a lot for the explanation.

closed 2015-07-31 by NilsL3