tum-vision/lsd_slam #129: collect2: error: ld returned 1 exit status, [../bin/videoStitch] Error 1

bl4ckc00k1e, 2015-08-04:
Hi, i have this issue, ROS Indigo+Ubuntu 14.04, Output in Verbose:

$rosmake lsd_slam -v
...
[rosmake-2] Starting >>> lsd_slam_viewer [ make ]

[ rosmake ] Last 40 linesd_slam_viewer: 2.4 sec ] [ 1 Active 40/42 Complete ]
{-------------------------------------------------------------------------------
main_stitchVideos.cpp:(.text+0x8f01): referencia a cv::String::deallocate()' sin definir main_stitchVideos.cpp:(.text+0x8f30): referencia acv::String::deallocate()' sin definir
main_stitchVideos.cpp:(.text+0x8fc9): referencia a cv::String::deallocate()' sin definir main_stitchVideos.cpp:(.text+0x8fdb): referencia acv::String::deallocate()' sin definir
CMakeFiles/videoStitch.dir/src/main_stitchVideos.cpp.o: En la función inlayVid(int, char**)': main_stitchVideos.cpp:(.text+0x9c62): referencia acv::imread(cv::String const&, int)' sin definir
main_stitchVideos.cpp:(.text+0x9c6a): referencia a cv::String::deallocate()' sin definir main_stitchVideos.cpp:(.text+0x9cb3): referencia acv::imread(cv::String const&, int)' sin definir
main_stitchVideos.cpp:(.text+0x9cbd): referencia a cv::String::deallocate()' sin definir main_stitchVideos.cpp:(.text+0x9d02): referencia acv::imread(cv::String const&, int)' sin definir
main_stitchVideos.cpp:(.text+0x9d0c): referencia a cv::String::deallocate()' sin definir main_stitchVideos.cpp:(.text+0x9d5a): referencia acv::imread(cv::String const&, int)' sin definir
main_stitchVideos.cpp:(.text+0x9d67): referencia a cv::String::deallocate()' sin definir main_stitchVideos.cpp:(.text+0x9db1): referencia acv::imread(cv::String const&, int)' sin definir
main_stitchVideos.cpp:(.text+0x9dbe): referencia a cv::String::deallocate()' sin definir main_stitchVideos.cpp:(.text+0xa629): referencia acv::String::allocate(unsigned long)' sin definir
main_stitchVideos.cpp:(.text+0xa654): referencia a cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)' sin definir main_stitchVideos.cpp:(.text+0xa661): referencia acv::String::deallocate()' sin definir
main_stitchVideos.cpp:(.text+0xa6c6): referencia a cv::String::allocate(unsigned long)' sin definir main_stitchVideos.cpp:(.text+0xa6e7): referencia acv::imshow(cv::String const&, cv::_InputArray const&)' sin definir
main_stitchVideos.cpp:(.text+0xa6f4): referencia a cv::String::deallocate()' sin definir main_stitchVideos.cpp:(.text+0xaec5): referencia acv::String::allocate(unsigned long)' sin definir
main_stitchVideos.cpp:(.text+0xaee8): referencia a cv::String::allocate(unsigned long)' sin definir main_stitchVideos.cpp:(.text+0xaf08): referencia acv::String::allocate(unsigned long)' sin definir
main_stitchVideos.cpp:(.text+0xaf28): referencia a cv::String::allocate(unsigned long)' sin definir main_stitchVideos.cpp:(.text+0xaf46): referencia acv::String::allocate(unsigned long)' sin definir
main_stitchVideos.cpp:(.text+0xb146): referencia a cv::String::deallocate()' sin definir main_stitchVideos.cpp:(.text+0xb1e2): referencia acv::String::deallocate()' sin definir
main_stitchVideos.cpp:(.text+0xb20e): referencia a cv::String::deallocate()' sin definir main_stitchVideos.cpp:(.text+0xb23f): referencia acv::String::deallocate()' sin definir
main_stitchVideos.cpp:(.text+0xb270): referencia a cv::String::deallocate()' sin definir CMakeFiles/videoStitch.dir/src/main_stitchVideos.cpp.o:main_stitchVideos.cpp:(.text+0xb33e): más referencias acv::String::deallocate()' sin definir a continuación
collect2: error: ld returned 1 exit status
make[3]: *** [../bin/videoStitch] Error 1
make[3]: se sale del directorio «/opt/ros/rosbuild_ws/lsd_slam/lsd_slam_viewer/build»
make[2]: *** [CMakeFiles/videoStitch.dir/all] Error 2
make[2]: se sale del directorio «/opt/ros/rosbuild_ws/lsd_slam/lsd_slam_viewer/build»
make[1]: *** [all] Error 2
make[1]: se sale del directorio «/opt/ros/rosbuild_ws/lsd_slam/lsd_slam_viewer/build»
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package lsd_slam_viewer written to:
[ rosmake ] /home/bl4ckc00k1e/.ros/rosmake/rosmake_output-20150804-193623/lsd_slam_viewer/build_output.log

[rosmake-2] Finished <<< lsd_slam_viewer [FAIL] [ 2.41 seconds ]

[ rosmake ] Halting due to failure in package lsd_slam_viewer.
[ rosmake ] Waiting for other threads to complete.

[rosmake-7] [ Build Terminated Thread Exiting ]

[rosmake-3] [ Build Terminated Thread Exiting ]

[rosmake-5] [ Build Terminated Thread Exiting ]

[rosmake-1] [ Build Terminated Thread Exiting ]

[rosmake-4] [ Build Terminated Thread Exiting ]

[rosmake-6] [ Build Terminated Thread Exiting ]

[rosmake-0] [ Build Terminated Thread Exiting ]

[ rosmake ] Results:

[ rosmake ] Built 41 packages with 1 failures.

[ rosmake ] Summary output to directory

[ rosmake ] /home/bl4ckc00k1e/.ros/rosmake/rosmake_output-20150804-193623

cipri-tom, 2016-02-01:
Hi. Have you found a solution in the end ?

cipri-tom, 2016-02-01:
Try moving the opencv2 folder (from #163)