tum-vision/lsd_slam #130: Errors while running ros_make lsd_slam

anuragajay, 2015-08-05:
Hi. When I run ros_make lsd_slam, I get the following error and I don't have any idea about this. One common line in the error is error: ‘monotonic_clock’ in namespace ‘std::chrono’ does not name a type.
"""
                   from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/util/globalFuncs.cpp:21:
  /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/Timestamp.h:78:2: error: ‘monotonic_clock’ in namespace ‘std::chrono’ does not name a type
  /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/Timestamp.h:81:15: error: ‘monotonic_clock’ in namespace ‘std::chrono’ does not name a type
  In file included from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/SlamSystem.h:28:0,
                   from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/SlamSystem.cpp:21:
  /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/Timestamp.h:78:2: error: ‘monotonic_clock’ in namespace ‘std::chrono’ does not name a type
  /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/Timestamp.h:81:15: error: ‘monotonic_clock’ in namespace ‘std::chrono’ does not name a type
  In file included from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/TimestampedObject.h:28:0,
                   from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/util/globalFuncs.h:24,
                   from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/Tracking/Sim3Tracker.cpp:25:
  /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/Timestamp.h:78:2: error: ‘monotonic_clock’ in namespace ‘std::chrono’ does not name a type
  /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/Timestamp.h:81:15: error: ‘monotonic_clock’ in namespace ‘std::chrono’ does not name a type
  make[3]: *** [CMakeFiles/lsdslam.dir/src/LiveSLAMWrapper.cpp.o] Error 1
  In file included from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/TimestampedObject.h:28:0,
                   from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/util/globalFuncs.h:24,
                   from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/Tracking/TrackingReference.cpp:25:
  /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/Timestamp.h:78:2: error: ‘monotonic_clock’ in namespace ‘std::chrono’ does not name a type
  /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/Timestamp.h:81:15: error: ‘monotonic_clock’ in namespace ‘std::chrono’ does not name a type
  In file included from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/TimestampedObject.h:28:0,
                   from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/util/globalFuncs.h:24,
                   from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/Tracking/SE3Tracker.cpp:25:
  /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/Timestamp.h:78:2: error: ‘monotonic_clock’ in namespace ‘std::chrono’ does not name a type
  /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/Timestamp.h:81:15: error: ‘monotonic_clock’ in namespace ‘std::chrono’ does not name a type
  In file included from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/TimestampedObject.h:28:0,
                   from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/util/globalFuncs.h:24,
                   from /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/DepthEstimation/DepthMap.cpp:31:
  /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/Timestamp.h:78:2: error: ‘monotonic_clock’ in namespace ‘std::chrono’ does not name a type
  /home/ajay/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/IOWrapper/Timestamp.h:81:15: error: ‘monotonic_clock’ in namespace ‘std::chrono’ does not name a type
"""

nguillermin, 2015-08-05:
Is this the full error message? Also, try adding four spaces in front of every line for proper code formatting on Github (this can be accomplished easily in Notepad++, Sublime Text, vi, and most other capable text editors).

anuragajay, 2015-08-05:
This was the entire error. I have also attached a screenshot.
[image]

anuragajay, 2015-08-05:
I figured that it was not the problem with package but with my g++ version

anuragajay, 2015-08-05:
But now, when I run play the rosbag file while running live_slam, I get the error:
    live_slam: malloc.c:3695: _int_malloc: Assertion `(unsigned long) (size) >= (unsigned long) (nb)' failed.

closed 2015-08-06 by anuragajay