tum-vision/lsd_slam #131: lsd_slam/pose topic

NilsL3, 2015-08-05:
Hello,

I have two questions concerning the lsd_slam/pose topic. Does the topic show the estimated position and orientation of the each frame provided by the camera or of the last keyframe?

My other question is if once lsd_slam has been initialized and the first keyframe has been grabbed, will the position at which the first keyframe was grabbed remain as position 0,0,0  (origin) during the entire camera movement?

I would appreciate your help a lot. Thanks.

weblucas, 2015-08-06:
Q1. The topic "lsd_slam/liveframes" provide the camera pose.

Q2. The first keyframe is in the origin with no rotation this means. Translation = (0,0,0) and orientation quaternion = (0,0,0,1)

anuragajay, 2015-08-06:
@weblucas 
I had some confusion regarding lsd_slam output. Does lsd_slam/liveframes shows estimated position and orientation of camera(w.r.t first image) in each of the images? If thats true, then would the final liveframe published by lsd_slam contain final estimated position and orientation of camera in each of the images?
If I understanding the above things wrong, can you explain me what is the output of lsd_slam and whether there is a way to get final estimated position and orientation of camera in each of the images.

Also, any guideline on how to save those liveframes as there is no "liveframes" message in ROS.

weblucas, 2015-08-06:
liveframes is a TOPIC. That is something like an information channel. The type of the message is keyframeMsg. Inside of this message there is a parameter called camToWorld that is the transformation from the camera coor system to the world system. My suggestion is to check the viewer code. There you can find how to save camera position. Again, The viewer shows the camera position in real time.

NilsL3, 2015-08-07:
Thanks for the help. I shall try a few things out and let you know how it works out. My other doubt would be, if the lsd_slam/liveframes topic provides camera pose, then what information does the lsd_slam/pose topic provide?

anuragajay, 2015-08-07:
@weblucas 
Sorry for repeated questions, but I am still somewhat confused. camToWorld transformation is published for every image. Is it right that if we have to find transform a pose from cam to world frame in a particular image, we use camToWorld published for that image? I am asking this because I am still not sure how to find camera pose using camToWorld transform?

weblucas, 2015-08-07:
The difference between liveframe and pose is the format.
liveframe topic provide a KeyframeMsg. KeyframeMsg has the intrinsics parameters as well and the camToWold is the serialization of a instance of Sophous Sim(3).  Sim(3) is a lie group with 7DoF. In other words is a fancy way to write a rigid body transformation with scale. the lsd_slam/pose only has extrinsic parameters using a quaternion and a vec3 translation. 

The easiest way to undestand how to use the lifeframe is the viewer code. or just use the pose.

anuragajay, 2015-08-07:
@weblucas
By viewer's code, do u mean main_viewer.cpp?

weblucas, 2015-08-07:
KeyFrameDisplay.cpp is where you can find the stuff

NilsL3, 2015-08-08:
Thanks. Indeed I found the camToWorld parameter in the KeyFrameDisplay.cpp. How could I get the 7DoF printed out for each image in the terminal or publish it on a separate topic?

weblucas, 2015-08-09:
you can use only the pose topic for this.

anuragajay, 2015-08-09:
@weblucas 
So, pose topic gives transformation from cam to world frame in PoseStamped format, right?

weblucas, 2015-08-09:
Yes

NilsL3, 2015-08-11:
Thanks for the responses. I have another question not exactly related to the topic of this issue. I am curious about what the constraints are and what the green and red lines mean, which are in between the keyframes in the viewer window?

weblucas, 2015-08-12:
They are the links(edges) used on the g2o for graph optimization.
http://ais.informatik.uni-freiburg.de/publications/papers/kuemmerle11icra.pdf
<http://ais.informatik.uni-freiburg.de/publications/papers/kuemmerle11icra.pdf>

SergioGarG, 2015-11-09:
Hi @weblucas , I've read the code and I'm comparing the data given by lsd_slam/pose and lsd_slam/liveframes topics. Pose is giving the pose and rotation of the camera (x, y, z and w, x, y, z). Should liveframes be giving the same (I know that 3 of the 7DoF are the pose x, y, z... but the other 4?)?

Any help will be welcome, thanks.

weblucas, 2015-11-09:
Liveframe give you more information. point cloud e other stuff. It is all
about the type of the message.

SergioGarG, 2015-11-10:
Ok, thank you very much

dk683, 2016-03-24:
I am sorry for asking a repeating question. Could I ask what the red and green links mean in the viewer window? Thank you for your help!

ank700, 2016-04-29:
Hello @weblucas , can you please see my problem #190 . I would like your help. Thanks

hashim19, 2016-07-13:
@weblucas the pose topic does not give accurate distance travelled or orientation made.... how is it correct then ?

ank700, 2016-07-14:
@hashim19 
 Since it is a monocular algorithm, you will not get pose in metric scale, if that is what you mean by not accurate.

hashim19, 2016-07-14:
@ank700 what do pose here tell us ? whats the values in the x,y and z direction ?