tum-vision/lsd_slam #132: Getting memory corruption while running data_slam

anuragajay, 2015-08-06:
When I run data_slam on my dataset, I get memory corruption error. I have attached the screenshot for the error. Also, for purpose of testing, if I use LSD_room_images on data_slam, which camera calibration file should I use? Is it in LSD_room_images.zip itself or somewhere else?

anuragajay, 2015-08-06:
Just an update. When I run dataset_slam with LSD_room Dataset, I get segmentation fault.
[image]

anuragajay, 2015-08-06:
Figured the error

closed 2015-08-06 by anuragajay

mihaibujanca, 2015-11-13:
@anuragajay what was the problem? How did you solve it?

kungfrank, 2018-03-10:
Is there any one know how to solve this error?