tum-vision/lsd_slam #136: install LSD-SLAM Failure

xiaokeme, 2015-08-11:
I try to install lsd-slam but I encountered some problem. I did the command line rosmake lsd_slam but I got errors during the compilation:  rosmake lsd_slam
 [ rosmake ] rosmake starting...                                                 
[ rosmake ] Packages requested are: ['lsd_slam']                                
[ rosmake ] Logging to directory /home/liuxunqian/.ros/rosmake/rosmake_output-20150812-030852
[ rosmake ] Expanded args ['lsd_slam'] to:
[]                                   
[ rosmake ] WARNING: The following args could not be parsed as stacks or packages: ['lsd_slam']
[ rosmake ] ERROR: No arguments could be parsed into valid package or stack names.

I use ROS fuerte+Ubuntu 12.04

Arseny-N, 2015-08-14:
`rosmake` does not see lsd_slam `stack.xml` file. Check the [ROS_PACKAGE_PATH](http://wiki.ros.org/ROS/EnvironmentVariables#ROS_PACKAGE_PATH) environment variable.

meirela, 2016-04-11:
For me it didn't help....
I've added my lsd_slam ROOT, the one that contains:
`LICENSE  lsd_slam_core  lsd_slam_viewer  README.md  stack.xml`

Now, my `ROS_PACKAGE_PATH ` looks like this (lsd_slam is the only package i've got) :
`/opt/ros/indigo/share:/opt/ros/indigo/stacks:/rosbuild_ws/package_dir/lsd_slam`

Where, the last path is lsd_slam's ROOT as specified above.

I'm still getting the same error message:
`[ rosmake ] rosmake starting...                                                                                                                                                                              
[ rosmake ] Packages requested are: ['lsd_slam']                                                                                                                                                             
[ rosmake ] Logging to directory /home/cvil/.ros/rosmake/rosmake_output-20160411-123115                                                                                                                      
[ rosmake ] Expanded args ['lsd_slam'] to:
[]                                                                                                                                                                
[ rosmake ] WARNING: The following args could not be parsed as stacks or packages: ['lsd_slam']                                                                                                              
[ rosmake ] ERROR: No arguments could be parsed into valid package or stack names.`

I am using ROS indigo+Ubuntu 14.04 btw.

blindnote, 2016-11-23:
@meirela remove lsd_slam , i.e. your ROS_PACKAGE_PATH should be like  
/opt/ros/indigo/share:/opt/ros/indigo/stacks:/rosbuild_ws/package_dir