tum-vision/lsd_slam #137: Doubts regarding camera calibration file.

anuragajay, 2015-08-12:
In OpenCv camera model, What are k1 k2 p1 p2? Are they first four distortion cofficient?
I am asking cause I am not sure what to use for them. My camera info is:

height: 240
width: 320
distortion_model: plumb_bob
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [159.99941228826285, 0.0, 160.5, 0.0, 159.99941228826285, 120.5, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [159.99941228826285, 0.0, 160.5, -0.0, 0.0, 159.99941228826285, 120.5, 0.0, 0.0, 0.0, 1.0, 0.0]

alejandronespereira, 2017-01-30:
Hi @anuragajay, I am facing the same doubt as you. Did you find the solution?

mfilipen, 2018-04-10:
Distortion coefficients = (k1, k2, p1, p2, k3)
k3 = 0
For more information: https://docs.opencv.org/3.1.0/dc/dbb/tutorial_py_calibration.html