tum-vision/lsd_slam #138: how to get LSD_room/cameraCalibration.cfg

xiaokeme, 2015-08-12:
where is the cameraCalibration.cfg of  dataset slam,please help me

weblucas, 2015-08-12:
the LSD_room.bag has inside the intrinsic parameters (camera_info). you only need to provide this in case of your camera is not publishing this topic.

xiaokeme, 2015-08-12:
yes! I have do it follow  README.md section 1,and it works well .but now I want to connect usb camera(it is an ordinary camera) to run “ live_slam  ”.
but I don't Know what does the “yourstreamtopic” mean.So I try to only use cameraCalibration.cfg of dataset slam to run the following order. but Not found the cameraCalibration.cfg.
rosrun lsd_slam_core live_slam /image:=<yourstreamtopic> /camera_info:=<yourcamera_infotopic>

if there is no cameraCalibration.cfg. how can I run “ live_slam  ” with an ordinary usb camera?

weblucas, 2015-08-12:
you need to read a bit about ROS concepts, to understand about topics. it will be very difficult to use this code without that.
You need to use a node to read your camera and publish the image_raw topic
http://wiki.ros.org/usb_cam

for calibration and generation of the camera_info file follow this tutorial.
http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration

good luck

xiaokeme, 2015-08-12:
thanks a lot.
Enjoy yourself！