tum-vision/lsd_slam #145: RGB Pointcloud

alexanderkoumis, 2015-08-22:
Hello, I am interested in retrieving RGB values from the keyframe somehow. I'm wondering if there is a debug option that I'm missing considering InputPointDense's color array of size 4. From what I can tell in the source it looks like RGB/RGBA images are converted to grayscale and are not used further, so I'm doubtful this has been implemented.

Not being too familiar with the system's internals, I'm wondering if every time buildImage is called to create the depth image there is some corresponding grayscale image kept in memory that is used to produce the intensity value at that point. My thought is that if this is so, a corresponding data structure of RGB images could be maintained and polled at this point instead in order to produce a colored point.

If this is not the case, another idea is to use TUM's DVO SLAM to generate the colored pointclouds, using the depth map generated by LSD SLAM in a new ROS message to simulate a RGB-D camera (I'm guessing this would require enabling LSD SLAM's option to regenerate the pointcloud for every image?).

alexanderkoumis, 2015-08-23:
Working on modifications in fork:

https://github.com/alexanderkoumis/lsd_slam/tree/rgb-pointcloud

amiltonwong, 2015-08-23:
Well done @alexanderkoumis , it is a good work worth following.
In https://github.com/alexanderkoumis/lsd_slam/tree/rgb-pointcloud
In your initial modification, I see some changes such as to append some pointer to rgbImage. As the final output is RGB point cloud. I think the Output3DWrapper should also be modified.

Best~
Milton

alexanderkoumis, 2015-08-23:
@amiltonwong thanks, have pushed completed modifications, point cloud looks pretty cool:
[image]

amiltonwong, 2015-08-24:
Great work, @alexanderkoumis . I had tried it and it works well.
One more thing may need to do similar modification for package lsd_slam_viewer if one want to save the mono point cloud into RGB point cloud

sunrentao, 2018-07-11:
hi, @alexanderkoumis ,I try your fork , but it always shows Segmentation fault(core dumped). Do you have any idea about this?  Meanwhile I try to solve it with #14 but it still doesn't work .
Thank your!

omarabid59, 2018-09-08:
For anyone that may be interested, I had a difficult time getting @alexanderkoumis RGB Pointcloud working on Ubuntu 16.04. For this reason, I've pushed a modified repository. Also added some additional code to output the live pointcloud frame to a publisher topic on ROS.

bespoke-code, 2018-09-09:
@omarabid59 if all is well, could you please open a PR and push the changes to Kevin George's repo? That way we can all have RGB pointclouds by cloning 1 repository :)