tum-vision/lsd_slam #151: Qt error, when install

XiaoshuiHuang, 2015-09-12:
[rosmake-0] Starting >>> mk [ make ]                                            
[rosmake-0] Finished <<< mk  No Makefile in package mk                          
[rosmake-1] Finished <<< rosbag  No Makefile in package rosbag                  
[rosmake-1] Starting >>> lsd_slam_viewer [ make ]                               
[ rosmake ] Last 40 linesd_slam_viewer: 6.3 sec ]    [ 1 Active 13/15 Complete ]
{-------------------------------------------------------------------------------
                   from /usr/include/qt5/QtCore/qobjectdefs.h:45,
                   from /usr/include/qt5/QtCore/qobject.h:47,
                   from /usr/include/QGLViewer/frame.h:30,
                   from /usr/include/QGLViewer/manipulatedFrame.h:26,
                   from /usr/include/QGLViewer/manipulatedCameraFrame.h:26,
                   from /usr/include/QGLViewer/camera.h:26,
                   from /usr/include/QGLViewer/qglviewer.h:26,
                   from /home/opendtam/openDTAM/OpenDTAM-ptam_poses/Cpp/build/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h:26,
                   from /home/opendtam/openDTAM/OpenDTAM-ptam_poses/Cpp/build/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.cpp:22:
  /usr/include/qt5/QtCore/qglobal.h:1079:4: error: #error "You must build your code with position independent code if Qt was built with -reduce-relocations. " "Compile your code with -fPIC or -fPIE."
  In file included from /usr/include/qt5/QtCore/qnamespace.h:45:0,
                   from /usr/include/qt5/QtCore/qobjectdefs.h:45,
                   from /usr/include/qt5/QtCore/qobject.h:47,
                   from /usr/include/QGLViewer/frame.h:30,
                   from /usr/include/QGLViewer/manipulatedFrame.h:26,
                   from /usr/include/QGLViewer/manipulatedCameraFrame.h:26,
                   from /usr/include/QGLViewer/camera.h:26,
                   from /usr/include/QGLViewer/qglviewer.h:26,
                   from /home/opendtam/openDTAM/OpenDTAM-ptam_poses/Cpp/build/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h:26,
                   from /home/opendtam/openDTAM/OpenDTAM-ptam_poses/Cpp/build/lsd_slam/lsd_slam_viewer/src/main_viewer.cpp:25:
  /usr/include/qt5/QtCore/qglobal.h:1079:4: error: #error "You must build your code with position independent code if Qt was built with -reduce-relocations. " "Compile your code with -fPIC or -fPIE."
  In file included from /home/opendtam/openDTAM/OpenDTAM-ptam_poses/Cpp/build/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h:26:0,
                   from /home/opendtam/openDTAM/OpenDTAM-ptam_poses/Cpp/build/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.cpp:22:
  /usr/include/QGLViewer/qglviewer.h:102:76: error: ‘Qt::WFlags’ has not been declared
  /usr/include/QGLViewer/qglviewer.h:103:97: error: ‘Qt::WFlags’ has not been declared
  /usr/include/QGLViewer/qglviewer.h:104:101: error: ‘Qt::WFlags’ has not been declared
  In file included from /home/opendtam/openDTAM/OpenDTAM-ptam_poses/Cpp/build/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h:26:0,
                   from /home/opendtam/openDTAM/OpenDTAM-ptam_poses/Cpp/build/lsd_slam/lsd_slam_viewer/src/main_viewer.cpp:25:
  /usr/include/QGLViewer/qglviewer.h:102:76: error: ‘Qt::WFlags’ has not been declared
  /usr/include/QGLViewer/qglviewer.h:103:97: error: ‘Qt::WFlags’ has not been declared
  /usr/include/QGLViewer/qglviewer.h:104:101: error: ‘Qt::WFlags’ has not been declared
  make[3]: *** [CMakeFiles/viewer.dir/src/PointCloudViewer.o] Error 1
  make[3]: *** Waiting for unfinished jobs....
  make[3]: *** [CMakeFiles/viewer.dir/src/main_viewer.o] Error 1
  make[3]: Leaving directory `/home/opendtam/openDTAM/OpenDTAM-ptam_poses/Cpp/build/lsd_slam/lsd_slam_viewer/build'
  make[2]: *** [CMakeFiles/viewer.dir/all] Error 2
  make[2]: Leaving directory `/home/opendtam/openDTAM/OpenDTAM-ptam_poses/Cpp/build/lsd_slam/lsd_slam_viewer/build'
  make[1]: *** [all] Error 2
  make[1]: Leaving directory `/home/opendtam/openDTAM/OpenDTAM-ptam_poses/Cpp/build/lsd_slam/lsd_slam_viewer/build'
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package lsd_slam_viewer written to:
[ rosmake ]    /home/opendtam/.ros/rosmake/rosmake_output-20150912-110826/lsd_slam_viewer/build_output.log
[rosmake-1] Finished <<< lsd_slam_viewer [FAIL] [ 6.33 seconds ]                
[ rosmake ] Halting due to failure in package lsd_slam_viewer. 
[ rosmake ] Waiting for other threads to complete.
[ rosmake ] Results:                                                            
[ rosmake ] Built 14 packages with 1 failures.                                  
[ rosmake ] Summary output to directory                                         

I used ubuntu 12.04, fuerte and all 'pwd' to ROS_PACKAGE_PATH. Do you know how to fix the problem coming from Qt?

XiaoshuiHuang, 2015-10-08:
I have solved this problem, please contact me if you need help

blankered, 2016-06-09:
Can you please post your solution?

XiaoshuiHuang, 2016-06-09:
It is the problem of OpenCV and Qt. 
My blog may help you. http://3dcrosssource.blogspot.com.au/2015/10/step-by-step-of-lsdslam-compile-in.html 
Another simple thing may help you fix the problem is to specify the QT version to version 4. 
use "qmake --version" to check your QT installation, and use QT_SELECT=qt4 to change QT environment.

Sorry, I am currently using Windows computer now, so I can answer you based on my memory. If there still exits any problem, I can check it in that Linux computer a bit far away from me.