tum-vision/lsd_slam #153: When will the code for two new papers extending from LSD-SLAM be released?

amiltonwong, 2015-09-22:
Hi, @JakobEngel ,

As the two new extended papers ("Large-Scale Direct SLAM with Stereo Cameras" and "Large-Scale Direct SLAM for Omnidirectional Cameras" ) are released recently, when will the code for these two new papers extending from LSD-SLAM be released?

Best~
Milton

mojovski, 2015-10-06:
I also really hope that LSD for stereo will be released. The paper looks amazing

sjobeek, 2015-12-06:
Very interested in the Stereo LSD-SLAM indeed - such great work.

lexvandersluijs, 2015-12-19:
My team is very interested in the stereo version as well, so we hope this will be released. Great work (again) !!

af-silva, 2015-12-29:
1+

skohlbr, 2016-01-20:
One more statement of interest in the stereo version here. A statement on future plans would be appreciated, even if it should say the new additions will not be available open source.

mojovski, 2016-02-29:
its published: https://github.com/raulmur/ORB_SLAM2

dk683, 2016-03-04:
Our group is also highly interested!

tsingjinyun, 2016-04-07:
can we collaborate on this Stereo lsd-slam code implement?

amiltonwong, 2016-04-07:
+join

mihaibujanca, 2016-04-07:
@tsingjinyun I'm in for that as well

tsingjinyun, 2016-06-23:
hi all，i come across some problem , https://github.com/tum-vision/lsd_slam/issues/203, this is very important and reference from http://www.ais.uni-bonn.de/papers/ICUAS_2015_Beul.pdf ,this give some advices on depth fusion. if anyone can figure this out ,please let me known...

antithing, 2016-11-22:
Definitely interested in the Omnidirectional approach! Will this code be made public as well? Thank you for all your work!

vien123, 2018-03-12:
I  also want to use Stereo LSD SLAM for my project, can you release it for us ?

EnriqueSolarte, 2018-11-10:
Hi everybody, 
May I ask you if someone knows about any the implementation (lsd-slam) for omnidirectional cameras is available?
thank you.