tum-vision/lsd_slam #154: Build on Raspberry Pi 2 Model B fails - boost mutex problem

gogobd, 2015-09-24:
Hello, every 01!

I am trying to compile LSD-SLAM on a Raspberry Pi 2. I got ROS installed and all dependencies mentioned are available now. I changed the CMakeList.txt in lsd_slam_core to get past some issues, but unfortunately I run into a problem I can't understand. Here's my session:

	Linux raspberrypi 4.1.6-v7+ #810 SMP PREEMPT Tue Aug 18 15:32:12 BST 2015 armv7l

	The programs included with the Debian GNU/Linux system are free software;
	the exact distribution terms for each program are described in the
	individual files in /usr/share/doc/*/copyright.

	Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
	permitted by applicable law.

	pi@raspberrypi ~ $ cd rosbuild_ws/
	pi@raspberrypi ~/rosbuild_ws $ source setup.bash
	pi@raspberrypi ~/rosbuild_ws $ cd package_dir/
	pi@raspberrypi ~/rosbuild_ws/package_dir $ git clone https://github.com/tum-vision/lsd_slam.git lsd_slam
	Cloning into 'lsd_slam'...
	remote: Counting objects: 490, done.
	remote: Total 490 (delta 0), reused 0 (delta 0), pack-reused 490
	Receiving objects: 100% (490/490), 1.11 MiB | 648 KiB/s, done.
	Resolving deltas: 100% (254/254), done.
	pi@raspberrypi ~/rosbuild_ws/package_dir $ cd lsd_slam/
	pi@raspberrypi ~/rosbuild_ws/package_dir/lsd_slam $ vi ./lsd_slam_core/CMakeLists.txt
	pi@raspberrypi ~/rosbuild_ws/package_dir/lsd_slam $ git diff
	diff --git a/lsd_slam_core/CMakeLists.txt b/lsd_slam_core/CMakeLists.txt
	index 758ce0b..8a4cfff 100644
	--- a/lsd_slam_core/CMakeLists.txt
	+++ b/lsd_slam_core/CMakeLists.txt
	@@ -35,11 +35,11 @@ gencfg()
	 # SSE flags
	 rosbuild_check_for_sse()
	 add_definitions("-DUSE_ROS")
	-add_definitions("-DENABLE_SSE")
	+#add_definitions("-DENABLE_SSE")
	 
	 # Also add some useful compiler flag
	 set(CMAKE_CXX_FLAGS
	-   "${CMAKE_CXX_FLAGS} ${SSE_FLAGS} -march=native  -std=c++0x"
	+   "${CMAKE_CXX_FLAGS} ${SSE_FLAGS} -std=c++0x"
	 ) 
	 
	 # Set source files
	pi@raspberrypi ~/rosbuild_ws/package_dir/lsd_slam $ rosmake lsd_slam
	[ rosmake ] rosmake starting...                                                                                                                                 
	[ rosmake ] Packages requested are: ['lsd_slam']                                                                                                                
	[ rosmake ] Logging to directory /home/pi/.ros/rosmake/rosmake_output-20150924-055837                                                                           
	[ rosmake ] Expanded args ['lsd_slam'] to:
	['lsd_slam_core', 'lsd_slam_viewer']                                                                                 
	[rosmake-0] Starting >>> catkin [ make ]                                                                                                                        
	[rosmake-0] Finished <<< catkin ROS_NOBUILD in package catkin
	 No Makefile in package catkin                                                                    
	[rosmake-0] Starting >>> genmsg [ make ]                                                                                                                        
	[rosmake-2] Starting >>> cpp_common [ make ]                                                                                                                    
	[rosmake-1] Starting >>> cmake_modules [ make ]                                                                                                                 
	[rosmake-3] Starting >>> rosgraph [ make ]                                                                                                                      
	[rosmake-0] Finished <<< genmsg ROS_NOBUILD in package genmsg
	 No Makefile in package genmsg                                                                    
	[rosmake-0] Starting >>> genlisp [ make ]                                                                                                                       
	[rosmake-0] Finished <<< genlisp ROS_NOBUILD in package genlisp.0 sec ] [ rosgraph: 0.0 sec ] [ genlisp: 0.0 sec ]                    [ 4 Active 2/42 Complete ]
	 No Makefile in package genlisp                                                                 
	[rosmake-3] Finished <<< rosgraph ROS_NOBUILD in package rosgraph
	 No Makefile in package rosgraph                                                              
	[rosmake-3] Starting >>> genpy [ make ]                                                                                                                         
	[rosmake-0] Starting >>> gencpp [ make ]                                                                                                                        
	[rosmake-1] Finished <<< cmake_modules ROS_NOBUILD in package cmake_modules
	 No Makefile in package cmake_modules                                               
	[rosmake-1] Starting >>> rospack [ make ]                                                                                                                       
	[rosmake-2] Finished <<< cpp_common ROS_NOBUILD in package cpp_common
	 No Makefile in package cpp_common                                                        
[... 132 lines cut ...]
	  make[3]: *** [CMakeFiles/lsdslam.dir/src/DataStructures/FrameMemory.cpp.o] Error 1
	  CMakeFiles/lsdslam.dir/build.make:123: recipe for target 'CMakeFiles/lsdslam.dir/src/SlamSystem.cpp.o' failed
	  make[3]: *** [CMakeFiles/lsdslam.dir/src/SlamSystem.cpp.o] Error 1
	  make[3]: Leaving directory '/home/pi/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
	  CMakeFiles/Makefile2:380: recipe for target 'CMakeFiles/lsdslam.dir/all' failed
	  make[2]: *** [CMakeFiles/lsdslam.dir/all] Error 2
	  make[2]: Leaving directory '/home/pi/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
	  Makefile:113: recipe for target 'all' failed
	  make[1]: *** [all] Error 2
	  make[1]: Leaving directory '/home/pi/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
	  /opt/ros/indigo/share/mk/cmake.mk:7: recipe for target 'all' failed
	-------------------------------------------------------------------------------}
	[ rosmake ] Output from build of package lsd_slam_core written to:                                                                   [ 1 Active 41/42 Complete ]
	[ rosmake ]    /home/pi/.ros/rosmake/rosmake_output-20150924-055837/lsd_slam_core/build_output.log
	[rosmake-1] Finished <<< lsd_slam_core [FAIL] [ 534.35 seconds ]                                                                                                
	[ rosmake ] Halting due to failure in package lsd_slam_core. 
	[ rosmake ] Waiting for other threads to complete.                                                
	[ rosmake ] Results:                                                                                                                                            
	[ rosmake ] Built 42 packages with 1 failures.                                                                                                                  
	[ rosmake ] Summary output to directory                                                                                                                         
	[ rosmake ] /home/pi/.ros/rosmake/rosmake_output-20150924-055837   

So something's wrong with a boost mutex lock, but I have no idea how I could fix that. It would be great to get it running, so any help would be very appreciated.

Gogo.

gogobd, 2015-09-25:
I'm upgrading my boost library from 1.49.0 to 1.59.0 to see what happens.

benjamincburns, 2016-01-03:
@gogobd Any update on this?

hca2014, 2016-01-09:
Can anyone provide the required changes to the CMakelists.txt file mentioned for RPi 2 compilation?
Thanks.

jerrykingnice, 2016-08-12:
hi gogobd,
how did you get this solved?I meet accross the same error.
thanks!

gogobd, 2016-08-12:
Unfortunately that was a while ago and I'm traveling right now, but I think I git that away by changing the version of the boost library. Sorry that I can't be more conclusive right now.

After many tries I got lsd-slam working without much graphical modules, especially without the point cloud viewer which is too heavy for my raspberry.

Hope that helps...

Gg
☆

SergioGarG, 2016-11-10:
Hi @gogobd ,  can you tell me how you solved the problem?

Thank you in advance :)