tum-vision/lsd_slam #155: How to extract output from lsd_slam - image/keyframe points etc

jsou, 2015-09-24:
I need to extract information from lsd_slam to use in another project. I know I can use the graph, keyframes, liveframes and pose topics, however how would I go about getting the coordinates of each 'point' and the image that it was found in? I thought this would be in the keyframe but I'm not sure how to interpret it. Any help would be appreciated, thanks.

jscatena88, 2015-09-24:
Take a look at this repo https://github.com/rbrth/framework/tree/master/slam_bridge. It is however an incomplete solution since updates to the keyframe graph will not be reflected

aartikanungo, 2015-10-27:
hello @jsou, i have same issue. i got the .ply file and csv file but i wanted to get coorsponding points of each image through which the cloud points are generated. if you get any solution regarding to this please let me know. Thank you

jsou, 2015-10-27:
I am working on a hack at the moment that gives the 2d points (as you see
on the debug window) and the associated image they were generated on. I'll
let you know

aartikanungo, 2015-11-05:
hello @jsou , did you get the solution?

dk683, 2016-05-22:
Hello. Is there any update/solution for this issue? Thank you!