tum-vision/lsd_slam #156: Couldn't find executable named viewer

artr0x, 2015-09-24:
When I try to run the viewer using `rosrun lsd_slam_viewer` ROS returns:
`[rosrun] Couldn't find executable named viewer below /opt/ros/indigo/share/lsd_slam/lsd_slam_viewer` 
Any ideas on what might be wrong? I followed the installation procedure as in the README. 

My `lsd_slam_viewer` folder is identical to the one in this git repository.

jscatena88, 2015-09-24:
lsd_slam shouldn't be found in your /opt/ros/ directory it should be in your catkin workspace. If you already have it in a workspace (probably in /home/[username]/my_proj or similar) make sure you have successfully compiled it and have sourced the setup.bash file

artr0x, 2015-09-25:
Thanks for the help! It turned out I had managed to skip installing one of the packages listed in 2.2 in the README. I've managed to get the minimal setup to work now.

I would like to try out live_slam using my laptop camera. Is that possible?  In that case how do I find out what `yourstreamtopic`and `yourcamera_infotopic` are supposed to be in `rosrun lsd_slam_core live_slam /image:=<yourstreamtopic> /camera_info:=<yourcamera_infotopic>`?

I'm very new to ROS, any pointers on where to look would be helpful

Anandembeddian, 2015-10-07:
Hey @artr0x , please follow this link for trying lsd_slam with laptop camera and it works...

http://www.hyperbolicthoughts.co.uk/2015/05/lsd-slam-install.html

can you please find my error and help me at #160 !!!!!

TungTNguyen, 2016-10-15:
Dear @Linusnie ,
I stuck in the same. After reading your comment, I opened roscore in another terminal, and re-installed every package. But the problem persists. Please describe your troubleshooting in more details.
LSD-SLAM is a very popular visual SLAM and I'm trying to apply it to a self driving car challenge. Any help would be much appreciated.

harshil2712, 2018-11-13:
> @TungTNguyen @Linusnie I ran into the same problem. But I I didn't skip installing any packages and rosmake seemed to be working fine. But still got the issue and indeed there is no executable named viewer below lsd_slam_viewer. Any suggestions would be appreciated.

This issue is probably because the package.xml file and the cmake file is not configured properly. Go through the detailed tutorial once for the installation.
I faced the same problem and now it is solved.

ouyangandy, 2019-03-26:
> Dear @Linusnie ,
> I stuck in the same. After reading your comment, I opened roscore in another terminal, and re-installed every package. But the problem persists. Please describe your troubleshooting [...]

did you fix it ? i meet same problem... any help would be appreciated.