tum-vision/lsd_slam #159: Getting debug depth window over IP

havi0019, 2015-10-05:
I am trying to remotely perform lsd slam with small computer/camera and have a another computer access the map, debug information.

when ssh'ing into the computer to run live slam I ran into this issue(on ubuntu 14.04):
(DebugWindow DEPTH:2747): Gtk-WARNING **: cannot open display:
Segmentation fault (core dumped)

The work-around to this problem was to disable displayDepthMap in util/settings.cpp

I have not looked at code much but would it be hard to be access the display depth map on another computer(or is there a way currently)? 

thx!

mihaibujanca, 2015-10-07:
@havi0019 You could try streaming the images directly to the second computer and run lsd_slam there rather than trying to access just some of the already processed info.

omarabid59, 2018-08-30:
You could try ssh'ing using the `ssh -Y hostname` flag.