tum-vision/lsd_slam #163: LSD SLAM Failures

yudifrank, 2015-10-22:
yudi@yudi-K46CM:~/rosbuild_ws$ rosmake lsd_slam
[ rosmake ] rosmake starting...                                                                
[ rosmake ] Packages requested are: ['lsd_slam']                                               
[ rosmake ] Logging to directory /home/yudi/.ros/rosmake/rosmake_output-20151022-213408        
[ rosmake ] Expanded args ['lsd_slam'] to:
['lsd_slam_viewer', 'lsd_slam_core']                
[rosmake-0] Starting >>> catkin [ make ]                                                       
[rosmake-0] Finished <<< catkin ROS_NOBUILD in package catkin        [ 1 Active 0/42 Complete ]
 No Makefile in package catkin   
[rosmake-0] Starting >>> genmsg [ make ]                                                       
[rosmake-3] Starting >>> cpp_common [ make ]                                                   
[rosmake-2] Starting >>> cmake_modules [ make ]                                                
[rosmake-3] Finished <<< cpp_common ROS_NOBUILD in package cpp_common
 No Makefile in package cpp_common
[rosmake-1] Starting >>> rosgraph [ make ]                                                     
[rosmake-0] Finished <<< genmsg ROS_NOBUILD in package genmsg
 No Makefile in package genmsg   
[rosmake-3] Starting >>> rostime [ make ]                                                      
[rosmake-0] Starting >>> genlisp [ make ]                                                      
[rosmake-1] Finished <<< rosgraph ROS_NOBUILD in package rosgraph
 No Makefile in package rosgraph
[rosmake-2] Finished <<< cmake_modules ROS_NOBUILD in package cmake_modules
 No Makefile in package cmake_modules
[rosmake-1] Starting >>> genpy [ make ]                                                        
[rosmake-2] Starting >>> gencpp [ make ]                                                       
[rosmake-3] Finished <<< rostime ROS_NOBUILD in package rostime
 No Makefile in package rostime
[rosmake-3] Starting >>> roscpp_traits [ make ]                                                
[rosmake-0] Finished <<< genlisp ROS_NOBUILD in package genlisp
 No Makefile in package genlisp
[rosmake-0] Starting >>> rospack [ make ]                                                      
[rosmake-3] Finished <<< roscpp_traits ROS_NOBUILD in package roscpp_traits
 No Makefile in package roscpp_traits
[rosmake-3] Starting >>> roscpp_serialization [ make ]                                         
[rosmake-2] Finished <<< gencpp ROS_NOBUILD in package gencpp
 No Makefile in package gencpp   
[rosmake-1] Finished <<< genpy ROS_NOBUILD in package genpy
 No Makefile in package genpy      
[rosmake-0] Finished <<< rospack ROS_NOBUILD in package rospack
 No Makefile in package rospack
[rosmake-2] Starting >>> rosparam [ make ]                                                     
[rosmake-0] Starting >>> roslib [ make ]                                                       
[rosmake-3] Finished <<< roscpp_serialization ROS_NOBUILD in package roscpp_serialization
 No Makefile in package roscpp_serialization
[rosmake-1] Starting >>> message_generation [ make ]                                           
[rosmake-3] Starting >>> message_runtime [ make ]                                              
[rosmake-2] Finished <<< rosparam ROS_NOBUILD in package rosparam
 No Makefile in package rosparam
[rosmake-2] Starting >>> rosmaster [ make ]                                                    
[rosmake-1] Finished <<< message_generation ROS_NOBUILD in package message_generation
 No Makefile in package message_generation
[rosmake-3] Finished <<< message_runtime ROS_NOBUILD in package message_runtime
 No Makefile in package message_runtime
[rosmake-1] Starting >>> roslang [ make ]                                                      
[rosmake-3] Starting >>> std_msgs [ make ]                                                     
[rosmake-0] Finished <<< roslib ROS_NOBUILD in package roslib
 No Makefile in package roslib   
[rosmake-0] Starting >>> rosbuild [ make ]                                                     
[rosmake-3] Finished <<< std_msgs ROS_NOBUILD in package std_msgs
 No Makefile in package std_msgs
[rosmake-1] Finished <<< roslang ROS_NOBUILD in package roslang
 No Makefile in package roslang
[rosmake-1] Starting >>> rosunit [ make ]                                                      
[rosmake-3] Starting >>> geometry_msgs [ make ]                                                
[rosmake-0] Finished <<< rosbuild ROS_NOBUILD in package rosbuild
 No Makefile in package rosbuild
[rosmake-2] Finished <<< rosmaster ROS_NOBUILD in package rosmaster
 No Makefile in package rosmaster
[rosmake-0] Starting >>> rosgraph_msgs [ make ]                                                
[rosmake-2] Starting >>> xmlrpcpp [ make ]                                                     
[... 83 lines cut ...]
  main_stitchVideos.cpp:(.text+0xaf28): undefined reference to `cv::String::allocate(unsigned long)'
  main_stitchVideos.cpp:(.text+0xaf46): undefined reference to `cv::String::allocate(unsigned long)'
  main_stitchVideos.cpp:(.text+0xb146): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0xb1e2): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0xb20e): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0xb23f): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0xb270): undefined reference to `cv::String::deallocate()'
  CMakeFiles/videoStitch.dir/src/main_stitchVideos.cpp.o:main_stitchVideos.cpp:(.text+0xb33e): more undefined references to `cv::String::deallocate()' follow
  collect2: error: ld returned 1 exit status
  make[3]: *** [../bin/videoStitch] Error 1
  make[3]: Leaving directory `/home/yudi/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
  make[2]: *** [CMakeFiles/videoStitch.dir/all] Error 2
  make[2]: Leaving directory `/home/yudi/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
  make[1]: *** [all] Error 2
  make[1]: Leaving directory `/home/yudi/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package lsd_slam_viewer written to:
[ rosmake ]    /home/yudi/.ros/rosmake/rosmake_output-20151022-213408/lsd_slam_viewer/build_output.log
[rosmake-0] Finished <<< lsd_slam_viewer [FAIL] [ 3.49 seconds ]                               
[ rosmake ] Halting due to failure in package lsd_slam_viewer. 
[ rosmake ] Waiting for other threads to complete.
[ rosmake ] Results:                                                                           
[ rosmake ] Built 41 packages with 1 failures.                                                 
[ rosmake ] Summary output to directory                                                        
[ rosmake ] /home/yudi/.ros/rosmake/rosmake_output-20151022-213408

mihaibujanca, 2015-10-22:
The build can't find your openCV. 

Just add `find_package(OpenCV REQUIRED)` and `${OPENCV_LIB}` to `target_link_libraries` and it rebuild

vadimostanin, 2015-11-25:
I had same issue,
I fixed it,
I had two includes and libs for opencv2 with differ version ( 2 and 3 ) at /usr/{include ; lib} and /usr/local/{include lib} respectively
When I renamed /usr/local/include/opencv2 to /usr/local/include/opencv2qwe or just delete it,
then 
cd ./lsd_slm/lsd_slam_viewer/build
make clean && make
successfuly compiled

cipri-tom, 2016-02-01:
@mihaibujanca that doesn't work if you have both openCV2 and openCV3. Do you know how we could specify which OpenCV to look for? I tried `find_package(OpenCV 3 REQUIRED)` and `find_package(OpenCV 3.0.0 REQUIRED)` with no success :disappointed: 

It seems by default it is selecting OpenCV2, as indicated by @vadimostanin's comment. His solution also worked for me, but I wish it could be done more cleanly :smile:

mihaibujanca, 2016-02-01:
@cipri-tom I'm pretty sure you will run into a number of problems if you will use OpenCV v3, as a number of functions from OpenCV 2 are deprecated or changed - so this might cause a load of problems you might not be aware of. My advice is to try and use OpenCV 2.4.8.

If you still want to use OpenCV3:
Download OpenCV3 and build it, **but don't install it**. If you install both opencv 2 and opencv 3, you will almost surely run into conflicts, with part of the files belonging to each version and some of them overwritten, depending on the order of the instalation.

In CMakeFiles.txt, instead of using find_package, manually add the include and lib path for opencv 
(e.g
 ```
set(OpenCV_INCLUDE_DIRS "path/to/opencv3/include")
set(OpenCV_LIBS "/path/to/opencv3/lib")
set(OpenCV_DIR "/path/to/opencv3/build")
```
)
Not sure if the above is exactly what you have to write, needs a bit of digging but should be something along those lines. If you use this, you should remove find_package(OpenCV).

Hope this helps

magicsn0w, 2016-10-10:
@vadimostanin 
OK !!! Thanks,You are right.It's useful for me.
The problem is solved.

TungTNguyen, 2016-10-16:
@mihaibujanca : I got into the error in order to generate the viewer binary in the lsd slam viewer:
{-------------------------------------------------------------------------------
         ^
  In file included from /home/tunggo/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/main_viewer.cpp:25:0:
  /home/tunggo/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h: In member function ‘std::__cxx11::string AnimationObject::toString()’:
  /home/tunggo/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h:136:26: error: invalid initialization of non-const reference of type ‘qreal& {aka double&}’ from an rvalue of type ‘qreal {aka double}’
     frame.getPosition(x,y,z);
                            ^
  In file included from /usr/include/QGLViewer/keyFrameInterpolator.h:31:0,
                   from /usr/include/QGLViewer/camera.h:26,
                   from /usr/include/QGLViewer/qglviewer.h:26,
                   from /home/tunggo/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h:26,
                   from /home/tunggo/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/main_viewer.cpp:25:
  /usr/include/QGLViewer/frame.h:188:7: note:   initializing argument 1 of ‘void qglviewer::Frame::getPosition(qreal&, qreal&, qreal&) const’
    void getPosition(qreal& x, qreal& y, qreal& z) const;
         ^
  /home/tunggo/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.cpp: In member function ‘virtual void PointCloudViewer::keyPressEvent(QKeyEvent*)’:
  /home/tunggo/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.cpp:327:44: error: invalid initialization of non-const reference of type ‘qreal& {aka double&}’ from an rvalue of type ‘qreal {aka double}’
          camera()->frame()->getPosition(x,y,z);
                                              ^
  In file included from /usr/include/QGLViewer/keyFrameInterpolator.h:31:0,
                   from /usr/include/QGLViewer/camera.h:26,
                   from /usr/include/QGLViewer/qglviewer.h:26,
                   from /home/tunggo/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h:26,
                   from /home/tunggo/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.cpp:22:
  /usr/include/QGLViewer/frame.h:188:7: note:   initializing argument 1 of ‘void qglviewer::Frame::getPosition(qreal&, qreal&, qreal&) const’
    void getPosition(qreal& x, qreal& y, qreal& z) const;
         ^
  CMakeFiles/viewer.dir/build.make:156: recipe for target 'CMakeFiles/viewer.dir/src/PointCloudViewer.cpp.o' failed
  make[3]: *** [CMakeFiles/viewer.dir/src/PointCloudViewer.cpp.o] Error 1
  make[3]: *** Waiting for unfinished jobs....
  CMakeFiles/viewer.dir/build.make:97: recipe for target 'CMakeFiles/viewer.dir/src/main_viewer.cpp.o' failed
  make[3]: *** [CMakeFiles/viewer.dir/src/main_viewer.cpp.o] Error 1
  make[3]: Leaving directory '/home/tunggo/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
  CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/viewer.dir/all' failed
  make[2]: *** [CMakeFiles/viewer.dir/all] Error 2
  make[2]: Leaving directory '/home/tunggo/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
  Makefile:127: recipe for target 'all' failed
  make[1]: *** [all] Error 2
  make[1]: Leaving directory '/home/tunggo/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
  /opt/ros/kinetic/share/mk/cmake.mk:7: recipe for target 'all' failed
-------------------------------------------------------------------------------}
Any idea, please?

cipri-tom, 2016-10-17:
@TungTNguyen these look like errors coming from unmatching C++ versions. Make sure you're using the correct C++ standard. I don't remember which one was for this project

TungTNguyen, 2016-10-17:
@cipri-tom : thank you for your diagnosis and suggestion. My gcc version is 5.4.0 on Ubuntu 16.04.2, while my lib qgl viewer is at its newest version 2.6.3 (more or less)
one stack over flow user and github user recommended to change the floats to qreals, but it didn't solved the error. 
What do you think?
Any help would be paid forward.

cipri-tom, 2016-10-24:
@TungTNguyen I thing that the only chance to get this to run is to reproduce their environment. So use Ubuntu 14.04 and ros-indigo. And on x86 (32 bits). This is important!! 
Then you may have some luck. But it's really difficult to get everything correctly!

mattdinhnguyen, 2016-11-11:
@TungTNguyen I ran into the same error: invalid initialization of non-const reference of type ‘qreal& {aka double&}’ from an rvalue of type ‘qreal {aka double}’ frame.getPosition(x,y,z);
How did you fix this? Thanks.

blindnote, 2016-11-23:
@mattdinhnguyen guess you're also installing on x86_64, I changed as below then it built through:

~/Workspace/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCLoudViewer.cpp:
326:  float x, y, z;  // change to —>   qreal x, y, z;

~/Workspace/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCLoudViewer.h:
135: float x, y, z;    //  change to —>   qreal x, y, z;

alejandronespereira, 2017-01-20:
This worked fine in a ubuntu 16.04 with Ros Kinect, thanks man!

Jyang772, 2017-04-29:
Following @blindnote 's suggestion and #237 it compiles for Ubuntu 16.04 with ROS Kinetic.

babyformula, 2017-06-26:
@cipri-tom Hi, I just tried in **Ubuntu 14.04 + indigo** and solved this problem perfectly, which may result from the conflicts among OpenCV of different version.
Here's the method: download and compile the OpenCV [2.4.8](https://github.com/Itseez/opencv/releases/tag/2.4.8) version, and `sudo make install` (after that u can also merge the latest version OpenCV by doing this again). 
And add `find_package(OpenCV 2.4.8 REQUIRED)` in both `lsd_slam_core/CMakeLists.txt` & `lsd_slam_viewer/CMakeLists.txt`

Young532042725, 2018-01-20:
@blindnote @mihaibujanca   Friends, may u give me some advice? rosmake lsd-slam with(16.04+ROS kinetic)  i have do these:
~/Workspace/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCLoudViewer.cpp:
326: float x, y, z; // change to —> qreal(or double) x, y, z;

~/Workspace/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCLoudViewer.h:
135: float x, y, z; // change to —> qreal(or double)x, y, z;
another problem occurs!  please!!!!  so desperate!!!
[image]

blindnote, 2018-01-21:
@Young532042725 check your opencv versions and references , looks like something is wrong there

ruoyuwangeel4930, 2018-03-24:
@Young532042725 I suppose you know Chinese. You can find a solution here.
https://blog.csdn.net/softimite_zifeng/article/details/78632653

ruoyuwangeel4930, 2018-03-24:
@Young532042725 I suppose you know Chinese, you can find a solution here.
https://blog.csdn.net/softimite_zifeng/article/details/78632653

ruoyuwangeel4930, 2018-03-24:
@yudiprtm I fixed the same problem as you by:
export LD_LIBRARY_PATH=$LD_LIBARARY_PATH:/usr/local/lib
sudo ldconfig