tum-vision/lsd_slam #165: rosmake lsd_slam problem

TimingSpace, 2015-10-31:
When i make the project
[ rosmake ] rosmake starting...                                                 
[ rosmake ] Packages requested are: ['lsd_slam']                                
[ rosmake ] Logging to directory /home/shikw/.ros/rosmake/rosmake_output-20151031-205636
[ rosmake ] Expanded args ['lsd_slam'] to:
['lsd_slam_viewer', 'lsd_slam_core'] 
[rosmake-0] Starting >>> lsd_slam_viewer [ make ]                               
[rosmake-1] Starting >>> lsd_slam_core [ make ]                                 
[rosmake-0] Finished <<< lsd_slam_viewer [SKIP] No rule to make target None     
[rosmake-1] Finished <<< lsd_slam_core [SKIP] No rule to make target None       
[ rosmake ] Results:                                                            
[ rosmake ] Built 2 packages with 0 failures.                                   
[ rosmake ] Summary output to directory                                         
[ rosmake ] /home/shikw/.ros/rosmake/rosmake_output-20151031-205636

agnivsen, 2015-11-10:
Did you find any solution to this? I'm also stuck with the same issue.

Are you using virtual box?

apullin, 2016-02-25:
I also had this problem, outside of a Virtualbox VM. Looking into the build log for lsd_slam_viewer, there was a failure related to libopencv. I agressively removed a bunch of packages, so I'm not sure which one actually resolved the error. Removing CUDA installs, and NVidia VisionWorks packages that were pulled in while using this computer to set up a Jetson TX1 board. Via apt, aptitude, and dpkg, all traces of libopencv were removed, and then reinstalled from Ubuntu main & ROS repos, and the error was gone.

tsingjinyun, 2016-04-28:
@apullin  i have make lsd_slam running ok on TX1,but the viewer doesn't show anything ,do you have meet this problem