tum-vision/lsd_slam #166: DebugWindow Dept window freeze after launch

kokasai, 2015-11-09:
I am very new to lsd slam and ros...
I am using LSD-SLAM with basler camera, but after I open lsd slam node, the DebugWindow Dept window just freeze after it appear.

-First I run roscore and then open camera node from Basler camera interface with https://github.com/shadow-robot/basler_camera 
-Then check, the camera seem to work fine.
-After that I run $rosrun lsd_slam_viewer viewer

Started QApplication thread
Started ROS thread
[ INFO] [1447058016.690150007]: lsd_slam_viewer started

appear on terminal

-then point cloud viewer open up it seem to work fine
-then I ran $rosrun lsd_slam_core live_slam image:=/camera/image_raw /camera_info:=/camera/camera_info

WAITING for ROS camera calibration!
Received ROS Camera Calibration: fx: 315.864044, fy: 311.883514, cx: 642.374084, cy: 441.329376 @ 1280x960
RECEIVED ROS camera calibration!
Started mapping thread!
Started  constraint search thread!
Doing Random initialization!
Started optimization thread 
started image display thread!
Done Random initialization!
init done 
opengl support available 

appear  on terminal
But the DebugWindow Dept window just freeze after it appear.

Even though I'm moving the camera there is nothing happen.. (w/o Tracking lost)

I don't know what seem to be the problem?

mihaibujanca, 2015-11-09:
+1 same issue, but I'll get tracking lost but not diverged after ~10 seconds

kokasai, 2015-11-12:
Did you also use Basler camera? I am thinking maybe is from the type of camera I used.

mihaibujanca, 2015-11-13:
@kokasai nope, I'm using an iPad camera. The camera is not the problem for me though, lsd_slam is losing track even for the datasets around frame 1500 for me + the depth window crashes instantly on live_slam.

closed 2016-01-27 by kokasai

maxhol, 2018-10-09:
+1 same issue did anyone solved it ? :)