tum-vision/lsd_slam #168: correct parameter choice

Seryogah, 2015-11-26:
I am trying to use LSD-SLAM with the camera provided in the 'Parrot Bebop Drone' but the results aren't that good as I hoped they would be. The 1080p@30FPS video stream from this specific camera is cropped to 640x480 and calibrated (I did this multiple times just to make sure). I know how to create the pointcloud from the keyframes and a few test-videos resulted in usefull points (but with lots of outliers).
Even though I tried multiple different cameras and parameters modified according to the notes, it never looked half as good as your presentations (e.g. the pose estimation in longer videos became completely wrong or tracking was lost after rotations).
I assume the problem lies in the cameras I tried (small fov of ~60-80) and thats why I would like to ask, if someone can push me in the right direction and find the correct parameters I need to make it work with the drone camera.

kiran4399, 2016-09-23:
@Seryogah .. Any updates on this issue? And BTW... FoV for Parrot Bebop's camera is pretty high.. 180 or so.. Did you check that?

kenisvery, 2017-01-19:
@Seryogah I'm also doing relevance project. Could you share the package on Github?

caap95, 2017-10-10:
Can you tell me how did you change those parameters? "KFUsageWeight", etc.  ?