tum-vision/lsd_slam #169: Get rid of ROS dependency

vranki, 2015-12-11:
ROS is supported only on some quite old Ubuntu versions. This makes it difficult to build lsd_slam on newer versions. The lsd_slam homepage says:

"However, ROS is only used for input (video), output (pointcloud & poses) and parameter handling; ROS-dependent code is tightly wrapped and can easily be replaced. "

I'd suggest replacing ROS with some widely available cross-platform library such as Qt. Qt supports video input (http://doc.qt.io/qt-5/cameraoverview.html) and other requirements. If possible, design the interfaces so that multiple alternatives can be supported.

mihaibujanca, 2015-12-11:
Actually ROS is supported on all versions of Ubuntu since 2010, including support for LTS12.04 (through Groovy), LTS14.04 (through indigo) and the latest, 15.04, through Jade. ROS are also having annual major releases. 

That being said, I don't enjoy ROS either, but I understand the choice. Qt is just a library, while ros is a toolkit

alexanderkoumis, 2015-12-11:
I disagree, the ROS interface makes it easy to integrate lsd_slam into an existing ROS-based stack and use it in conjunction any other ROS packages that use PCL pointclouds with [lsd_slam_to_pcl](https://github.com/alexanderkoumis/lsd_slam_to_pcl). Try [my fork](https://github.com/alexanderkoumis/lsd_slam) for a version that is easier to build with catkin (ROS' build tool) on ROS Indigo and Jade. ROS integration lets me use lsd_slam like a library (my fork has an option to disable the depth display) and would really dislike seeing it built into a standalone application with QT.

nickponline, 2016-02-19:
How big an under-taking would this be? I'm quite interested in a ROS-free version.

Seryogah, 2016-03-14:
LSD-SLAM only depends on ROS for the communication (cam<->lsdslam, lsdslam<->visualizer). If you don't care about the visualization, all you have to do is to write your own class for reading from images/video files. There are also some forks that are usable under windows and they don't use ROS.

ningt, 2016-03-18:
@nickponline I have successfully run a ROS-free version lsd-slam on my mac xcode. It uses a light-weight and open source framework [Pangolin](https://github.com/stevenlovegrove/Pangolin) for visualisation. You can check it out [here](https://github.com/ningt/lsd_slam_xcode).

LiangliangGuo, 2017-09-26:
Hi Seryogah!
I want to know how to run LSD_SLAM on windows,can you help me ?
many thanks

9friday, 2018-04-21:
@LiangliangGuo
I know that I am late(pretty late) to the party but maybe this could help : https://github.com/williammc/lsd_slam 
I haven't run it though. I am trying to get LSD-SLAM to run from a live videostream on Ubuntu 17.10 and just came across the above repo.
Cheers :grinning:

9friday, 2018-04-27:
I tried compiling on a different laptop.
On executing 'sudo make', the following shows up:

`
[  4%] Building CXX object CMakeFiles/lsd_slam.dir/depth_estimation/depth_map.cc.o
In file included from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/depth_estimation/depth_map.h:28:0,
                 from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/depth_estimation/depth_map.cc:21:
/home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:72:31: error: ‘SE3Group’ is not a class template
 extern template class Sophus::SE3Group<float, 0>;
                               ^~~~~~~~
/home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:72:31: error: ‘SE3Group’ in namespace ‘Sophus’ does not name a type
/home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:73:31: error: ‘SE3Group’ is not a class template
 extern template class Sophus::SE3Group<double, 0>;
                               ^~~~~~~~
/home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:73:31: error: ‘SE3Group’ in namespace ‘Sophus’ does not name a type
/home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:75:31: error: ‘Sim3Group’ is not a class template
 extern template class Sophus::Sim3Group<float, 0>;
                               ^~~~~~~~~
/home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:75:31: error: ‘Sim3Group’ in namespace ‘Sophus’ does not name a type
/home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:76:31: error: ‘Sim3Group’ is not a class template
 extern template class Sophus::Sim3Group<double, 0>;
                               ^~~~~~~~~
/home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:76:31: error: ‘Sim3Group’ in namespace ‘Sophus’ does not name a type
In file included from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/model/frame_pose_struct.h:23:0,
                 from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/model/frame.h:26,
                 from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/depth_estimation/depth_map.cc:31:
/home/username/Downloads/lsd_slam_noros-master/lsd_slam/global_mapping/g2o_type_sim3_sophus.h: In member function ‘virtual bool lsd_slam::EdgeSim3::setMeasurementData(const double*)’:
/home/username/Downloads/lsd_slam_noros-master/lsd_slam/global_mapping/g2o_type_sim3_sophus.h:96:20: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
   Eigen::Map<const g2o::Vector7d> v(m);
                    ^~~
/home/username/Downloads/lsd_slam_noros-master/lsd_slam/global_mapping/g2o_type_sim3_sophus.h:96:33: error: template argument 1 is invalid
   Eigen::Map<const g2o::Vector7d> v(m);
                                 ^
/home/username/Downloads/lsd_slam_noros-master/lsd_slam/global_mapping/g2o_type_sim3_sophus.h:96:38: error: invalid conversion from ‘const double*’ to ‘int’ [-fpermissive]
   Eigen::Map<const g2o::Vector7d> v(m);
                                      ^
/home/username/Downloads/lsd_slam_noros-master/lsd_slam/global_mapping/g2o_type_sim3_sophus.h:97:38: error: no matching function for call to ‘Sophus::Sim3<double>::exp(int&)’
   setMeasurement(Sophus::Sim3d::exp(v));
                                      ^
In file included from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:23:0,
                 from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/depth_estimation/depth_map.h:28,
                 from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/depth_estimation/depth_map.cc:21:
/usr/local/include/sophus/sim3.hpp:439:35: note: candidate: static Sophus::Sim3<Scalar_> Sophus::Sim3<Scalar_, Options>::exp(const Tangent&) [with Scalar_ = double; int Options = 0; Sophus::Sim3<Scalar_, Options>::Tangent = Eigen::Matrix<double, 7, 1, 0, 7, 1>]
   SOPHUS_FUNC static Sim3<Scalar> exp(Tangent const& a) {
                                   ^~~
/usr/local/include/sophus/sim3.hpp:439:35: note:   no known conversion for argument 1 from ‘int’ to ‘const Tangent& {aka const Eigen::Matrix<double, 7, 1, 0, 7, 1>&}’
CMakeFiles/lsd_slam.dir/build.make:62: recipe for target 'CMakeFiles/lsd_slam.dir/depth_estimation/depth_map.cc.o' failed
make[2]: *** [CMakeFiles/lsd_slam.dir/depth_estimation/depth_map.cc.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/lsd_slam.dir/all' failed
make[1]: *** [CMakeFiles/lsd_slam.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
`

This is again on a 64-bit laptop running Ubuntu 17.10.
Cheers :grinning: