tum-vision/lsd_slam #172: LSD SLAM results with datasets provided in http://vision.in.tum.de/research/vslam/lsdslam

lim0606, 2016-01-05:
Hi, everyone. 

I would like to reproduce the results provided in http://vision.in.tum.de/research/vslam/lsdslam. 

I ran `rosbag play <bag file>` with provided `.bag` files as illustrated in the quickstart (I installed with openFabMap). However, the results are generally not as good as the video clip described. In the case of the eccv sequence, the result is really bad. 

I didn't change any parameters yet. 

Is there anyone who can help me to solve this issue? 

Thank you

Jaehyun

lim0606, 2016-01-06:
As I can found in other issues, I ran `rosrun lsd_slam_core dataset_slam ...` instead of `live_slam` to use camera calibration files. The results with the camera configuration files are definitely better than without them. 

However, the results are still not as good as the video clips described. I especially tested the ECCV sequence with many different combinations of parameters, but they didn't work well. The system seems like we have to filter out some outliers for better performance.

Jaehyun.

jzubizarreta, 2017-05-05:
I have the same issues. The pointcloud with their database is noisy.
It is like the graph optimization is not working and many pointclouds are badly overlapped.
Did it happen the same to you? Did you solve it?

Thanks in advance.