tum-vision/lsd_slam #175: Depth map / pose with metric units.

richaagrawal, 2016-01-31:
We are interested in knowing the pose in metric units, say metres for our application. LSD SLAM is scale-less but the scene has an object dimension so we can use that to get the scale factor. The issue is incorporating the scale factor in the algorithm. 

It is similar to the issue here - https://github.com/tum-vision/lsd_slam/issues/48 - I am having issues adding it to the pose-graph optimization as suggested by Jakob Engel. Has anyone had any luck with adding scale to LSD, either to the pose-graph or directly to SLAM? 

Thanks.

dk683, 2016-05-06:
Hello @richaagrawal. Our group is also trying to achieve a similar goal (using LiDAR data).
Were you able to correctly incorporate the scale factor in the algorithm?

richaagrawal, 2016-05-09:
Nope, we weren't successful in doing so. For now, we are estimating scale through other means and scaling the final translation accordingly.

ank700, 2016-05-25:
@richaagrawal , I also want to set the scale value by calculating it myself. Can you please tell me where do you set the scale value in the source code? 
Also, what sensors do you use to calculate the scale? 
Thanks

srikar8, 2018-06-27:
@ghost @richaagrawal 
hello,
any update on the scale factor.