tum-vision/lsd_slam #191: TX1 has something wrong with lsd_slam_viewer?

tsingjinyun, 2016-04-28:
I have sucess make lsd_slam on Nvida Tx1,and could run example and my dasaset ,but  the lsd_slam_viewer just show a dark window and can't show any pointcloud? do any one come over this problem, thanks for any given suggestion!!!

sandmen, 2016-04-28:
Hi tsingjinyun,
currently working also on TX1 and SLAM.
I will setup today the lsd_slam. Are there some "special" settings, changes needed ?

By the way, the RTAB-MAP is working fine for me on the TX1.

tsingjinyun, 2016-04-29:
If you meet any questions,write here i will give my solution as far as i can,you may change the CMakelist to suit arm architecture

amiltonwong, 2016-04-29:
I had only Tk1, does it also suitable for LSD-SLAM?

sandmen, 2016-04-30:
Hmm, didn't find currently where the lsd-slam is define the videostab.so.2.4.8.
On TX1 we have 2.4.12, I need to change this some where... in Cmake or make.

sandmen, 2016-05-01:
Ok, sucking now on g2o/core/base_vertex.h files....
Any hints

tsingjinyun, 2016-05-05:
@amiltonwong ，I think it maybe ok,it just different in CPU,TK1 &Tx1 cpu core is just 4 ,this maybe a problem

tsingjinyun, 2016-05-05:
@sandmen did you change lsd_slam/lsd_slam_core/CMakeLists? about line 44,
"${CMAKE_CXX_FLAGS} ${SSE_FLAGS} -march=armv8-a  -std=c++0x"
and any other thing you change it ?

sandmen, 2016-05-05:
@tsingjinyun 
yes, did it. The main problem was, that the g2o lib was not installed :-(
And I need to disable SSE !! Do you also remove the "add_definitions("-DENABLE_SSE")" ?

tsingjinyun, 2016-05-06:
@sandmen yes,need to disable SSE(which is intel X86 architecture),so it seem that u didn't install g2o,try  apt-cache search ros-indigo-libg2o ,can u find it?
then run this commond sudo apt-get install ros-indigo-libg2o.

sandmen, 2016-05-06:
yes, install the bin. I try before to compile from source the g2o.
I try to use the ZED 3D cam with ld_slam, but currently hanging on calibration....

closed 2016-05-08 by tsingjinyun

dcs0002, 2016-06-17:
How did you resolve the black screen on the lsd_slam_viewer. 

I have everything running fine on TK1 and the viewer shows that it is receiving keyframes but the viewer itself remains black.

tngan, 2016-06-26:
@amiltonwong Did you successfully build this package on TK1 ?

tianwailaike, 2016-09-23:
@tsingjinyun Have you solved this problem, I also encountered this problem.The lsd_viewer is totally black and couldn't show any pointcloud.

migtissera, 2018-11-04:
Having the same issue on TX2 - any of you managed to resolve this issue?